km4ack/pi-build #180: update gpsd to latest version?

km4ack, 2020-11-26:
https://gitlab.com/gpsd/gpsd/-/blob/master/NEWS

This has potential to get rid of some of the problems experienced with 3.17 which is the latest in the pi repository as of NOV 2020

KC3LKV, 2020-12-10:
CONKY seems to be failing to start sometimes at reboot, and commands like gpsmon -n won't run with gpsd.  Upon troubleshooting, these appear to be linked.  For some reason, gpsd is interfering with conky.  I am looking at configuring my build to change the startup order between them to see if it helps.  However, as a interim action, I have found that if  CONKY doesn't start, killing the GPSD process and removing the socks for it, restarts CONKY.  (sudo killall gpsd, sudo killall -9 gpsd, sudo rm /run/gpsd.sock).  I am also able to access gpsmon displays.

km4ack, 2020-12-10:
This video should help for the time being. https://youtu.be/YIIC-iT6H-s

km4ack, 2021-03-15:
This issue was resolved in BAP 3.0.7. Once you upgrade to 3.0.7 or later, use the BAP update script and re install Conky.

closed 2021-03-15 by km4ack