km4ack/pi-build #187: Pat Error after BAP

kj7dmv, 2020-12-09:
Fresh BAP after installing the 12/2 Buster

I've made changes to direwolf.config and pat configure per instructions from the developer.

Trying to run the test version of pat that includes forms. I am trying to run from terminal.  I've downloaded the pat.rpi provided by the developer. Running using a headless VNC connection to my pi 3. I use pat-menu on a separate boot SD card and it works correctly. 

I start direwolf with -p -t 0 and that is fine. In a new tab I do 

./pat.rpi http

and I get 

2020/12/09 14:38:01 Starting HTTP service (0.0.0.0:5000)...
2020/12/09 14:38:01 listen tcp 0.0.0.0:5000: bind: address already in use

I get the same error if I start fresh and use pat-menu.

suggestions?

km4ack, 2020-12-09:
This isn't an issue with build a pi. You are trying to start Pat twice. Pat starts at boot as a system service when the Pi boots. Before starting your beta test or running any commands like direwolf, run "sudo killall pat" This will stop the Pat service until the pi is rebooted. You will also not be able to use Pat Menu for the beta test as it isn't configured to run with the beta version of Pat. You will need to start each needed service (direwolf, rig control, pat, etc) manually.

closed 2020-12-09 by km4ack

kj7dmv, 2020-12-10:
THATS what I was trying to remember! Thanks.