km4ack/pi-build #188: Feature Request - Add Kappanhang

blackwellj, 2020-12-12:
Please can you add in kappanhang? It adds support for wireless control of the IC705

https://github.com/nonoo/kappanhang

arcreigh, 2021-08-31:
I 100% agree with this. kappanhang is a lifesaver for me. 
Was having wicked USB induced noise and this little go app saved me and helped me make a nice DX call to Italy from WV.

km4ack, 2021-09-01:
Have you gotten this to successfully work with direwolf or ardopc?

arcreigh, 2021-09-01:
I have not as I just got my general and am just learning the software. That being said Kappanhang uses rigctl for cat control and creates a pulseaudio device for transmit and receive audio. There are additional command flags for software that cant handle those methods and prefer a serial device. My end goal is to be able to use pat winlink through kappanhang but I have just not quite found any guides on anyone doing so. In theory this should work with any software that supports hamlib rigctl, and I have tested this with wsjt-x and js8call. Those are confirmed to be working.

km4ack, 2021-09-01:
Yep. I have js8 & wsjtx working as well. No dice on direwolf or ardop. I created an [issue ticket](https://github.com/nonoo/kappanhang/issues/18) with the kappanhang developer this morning to see if he could advise. I was able to create a virtual sound card and pipe the incoming signal from kappanhang-705 to the virtual card which allowed direwolf to receive and decode APRS packets. Just no luck with TX.

km4ack, 2022-03-28:
A note that some progress has been made on this. PTT is working but we haven't gotten audio from ARDOP to the radio yet. See [this post](https://groups.io/g/KM4ACK-Pi/topic/pat_menu_with_kappanhang/89784796?p=,,,20,0,0,0::recentpostdate/sticky,,,20,2,0,89784796,previd%3D9223372036854775807,nextid%3D1648036403644636052&previd=9223372036854775807&nextid=1648036403644636052) for more details.

SpudGunMan, 2022-10-31:
FYI I do think wfview would be a more updated project here