km4ack/pi-build #192: List of Ham Radio apps

km4ack, 2020-12-22:
Just putting this here for future reference

[List of Apps](https://www.raspberryconnect.com/raspbian-packages/39-raspbian-hamradio)

km4ack, 2021-02-28:
A few additional resources:
https://www.raspberryconnect.com/raspbian-packages/39-raspbian-hamradio

https://orari.or.id/index.php/2019/09/02/setup-raspberry-pi-for-ham-radio/

https://photobyte.org/installing-upgrading-cqrlog-on-a-ras

http://www.w4wta.com/amateur_radio_software_w4wta.htm

kelvin0mql, 2022-07-05:
I might suggest that **every** time some new thing gets added to the script, that the README.md at the root of this project get at least a one-line addition.

I was impressed that there's a thing on the desktop that... uh... says stuff that's live/pertinent.

It took me waaaaaaaay too long to figure out that's because of "conky".

Frankly, "conky" is not an immediately intuitive name for a thing.

This might all be covered in a YouTube vid, but that is insufficient. I can't Ctrl-F for "desktop" or "status" or "conky" in a video.

And, yes, I am indeed volunteering to pitch in with keeping the glossary-of-stuff-this-can-install if that's what's necessary to make this happen. With that, off I go to find out what it means that under my callsign it says
Gridsquare     Some err
...but nowhere in that "oh, that's cool, I wonder what THAT is" does it say "conky" (which is what started all this).