km4ack/pi-build #194: after updating direwolf 1.5 to 1.6 ptt ic 7100 isn't working for packet 2m

RonMar26, 2020-12-27:
After updating direwolf 1.5 to 1.6 in build-a-pi 3.0.5 my ic7100 is not working anymore on the PTT function. All the functions in the packet ax25 on 2m are working accept the PTT is not reacting. I have looked in the direwolf.conf file but there is nothing to find what have been changed.
I hoop that there is someone who can give me a hint how to fix this problem.
PD7RON

km4ack, 2021-01-03:
Which sound card are you using?

RonMar26, 2021-01-03:
Jason I am using the soundcard in my icom 7100

km4ack, 2021-01-03:
You will need to add this PTT statement to your direwolf.conf file:
`PTT RIG 2 localhost:4532`

This assumes that you are using FLRIG for rig control. In the direwolf.conf file look around line 186 for other PTT examples and try adding the line above just below the examples in the conf file.

73, de KM4ACK

RonMar26, 2021-01-03:
Ok Jason I am going to try this. I’ll let you know if it fixed the problem. 73 Ron PD7RON Verzonden vanuit Mail voor Windows 10 Van: KM4ACKVerzonden: zondag 3 januari 2021 18:20Aan: km4ack/pi-buildCC: RonMar26; AuthorOnderwerp: Re: [km4ack/pi-build] after updating direwolf 1.5 to 1.6 ptt ic 7100 isn't working for packet 2m (#194) You will need to add this PTT statement to your direwolf.conf file:PTT RIG 2 localhost:4532This assumes that you are using FLRIG for rig control. In the direwolf.conf file look around line 186 for other PTT examples and try adding the line above just below the examples in the conf file.73, de KM4ACK

RonMar26, 2021-01-03:
Jason I disabled the line with
PTT RIG 370 /dev/ttyUSB0 RTS
I added PTT RIG 2 localhost: 4532
And now it is working again
Thanks for your help.

73 Ron
PD7RON

closed 2021-01-03 by km4ack