km4ack/pi-build #195: SSIDs truncated when selecting WiFi network

dpsm64, 2020-12-30:
The list of networks that comes up when you're first starting build-a-pi seems to truncate SSIDs that have spaces or underscores in them. The attached shows my network as "The" when it's really "The Hack Shack 5G". Not a problem for selecting networks -- I was still able to figure out which one was mine and log in no problem. But it might be something to fix when you get a chance.

Thanks for the script, very nicely done. Donation inbound. 73 DE N7DPM

wdbrady, 2021-01-08:
I had the same issue, but also have multiple entries with the same "prefix".   I assume this has to do with the fact that I am using an Eero mesh network.  In the attached screen print, my network starts with El and you can see I have multiple entries in addition to my neighbors.   
[image]

As always, great job on this project!

km4ack, 2021-02-17:
slated for 3.0.6 release now in beta

closed 2021-02-17 by km4ack