km4ack/pi-build #196: Invalid desktop entry file: '/usr/share/applications/build-a-pi.desktop

imarunner2, 2020-12-31:
I opened Build-a-Pi. It said a newer version was available. The update didn't finish and I get this error when I try to re-run Build-a-Pi... also the cool Build-a-Pi  wallpaper is gone. 

Invalid desktop entry file: '/usr/share/applications/build-a-pi.desktop

I suspect the first part of update removes this file?

km4ack, 2020-12-31:
Run this command and post the results please:
`ls ~/pi-build`

imarunner2, 2020-12-31:
Sorry, I've already re-installed so all looks good now.  I did look to see
what was in pi-build and i believe the update script had been deleted when
it failed to complete and that's what was causing the issue when I tried
rerunning it... but I wouldn't swear to it.

pi@buildapi:~/pi-build $  ls ~/pi-build
bap-wallpaper.jpg  changelog  conky     diagnostic  logo.png   update
build-a-pi         config     cron.tmp  functions   README.md

73,
KB9LNS

closed 2020-12-31 by km4ack