km4ack/pi-build #198: Conky Clock Freeze Solution?

km4ack, 2021-01-03:
Can the get-grid script be modded to write to a simple text file and then pull that data into Conky from the text file? Look at [this post](https://groups.io/g/KM4ACK-Pi/message/2558) If so, it may be possible to eliminate the freezing clock issue. This may also fix the problem of Conky not starting if there are other GPS issues that occur.

starbasessd, 2021-01-03:
It could be as simple as
crontab -e:
*/10 * * * * /home/pi/grid > /home/pi/gridsquare  # run grid check every 10 seconds

and change the .conkyrc file to display the /home/pi/gridsquare file instead of running /home/pi/grid directly.

Ditto for any other  program run from .conkyrc. Rather than make conky wait for results, run the program elsewhere and just have conky display the results. That way if a program crrashes it doesn't take everything else down.

Other than time, I don't see anything else there that needs to be a real 'live' update

starbasessd, 2021-01-03:
In fact it is.
In .conkyrc
Replace line:
${font Arial:bold:size=18}${color White}Gridsquare ${alignr}${color Yellow}${execi 25 /home/pi/grid | cut -c1-8}
with
${font Arial:bold:size=18}${color White}Gridsquare ${alignr}${color Yellow}${exec cat /home/pi/grid.coord | cut -c1-6}
In
sudo crontab -e
add line
*/3 * * * * /home/pi/grid > /home/pi/grid-coord
Change the */3 (which checks it every 3 minutes) to whatever you want.
Should pick up the changes within 3 minutes (or whatever you have set) and no more delays caused by conky reading the /home/pi/grid program.

km4ack, 2021-02-17:
implemented in 3.0.6 beta

closed 2021-02-17 by km4ack