km4ack/pi-build #200: Use Bluetooth GPS from tablet

vbursucianu, 2021-01-13:
Personnaly i dont see the point in buying an usb gps dongle to plug on the Raspberry pi if my tablet already has a gps and can share it's data over bluetooth using various apps on the playstore.
This is ideal since it cost less and has less hardware dangling aroung the raspberry pi. And i am pretty sure that most hams will either use a cell phone or a tablet to connect on their raspberry pi.
An improvement to the installer could be the support of GPS using bluetooth instead of USB.

vbursucianu, 2021-02-16:
Finally!
https://www.youtube.com/watch?v=rkmQxvT8BpQ

closed 2021-02-16 by vbursucianu

km4ack, 2021-02-16:
Took me a few weeks to get around to it but the video solves the issue for Android and iPhone users. The GPSupdate tool will make it super simple to implement and take away the manual configuration. Should be coming in just a couple of more weeks. Enjoy!