km4ack/pi-build #201: Bug -

vbursucianu, 2021-01-14:
I've managed to install everything (without GPS and RTC) and i can't install the FLDIGI because it crash at compilation with this error message:
```
/usr/bin/ld: dialogs/fldigi-confdialog.o: file not recognized: file truncated
collect2: error: ld returned 1 exit status
make[3]: *** [Makefile:3154: fldigi] Error 1
make[2]: *** [Makefile:8643: all-recursive] Error 1
make[1]: *** [Makefile:2024: all] Error 2
make: *** [Makefile:469: all-recursive] Error 1
/usr/bin/ld: dialogs/fldigi-confdialog.o: file not recognized: file truncated
collect2: error: ld returned 1 exit status
make[3]: *** [Makefile:3154: fldigi] Error 1
make[2]: *** [Makefile:8643: install-recursive] Error 1
make[1]: *** [Makefile:8974: install] Error 2
make: *** [Makefile:469: install-recursive] Error 1
```
I am using a Raspberry Pi 3B+ with the latest freshly installed rasbian version OS released on January 11th 2021.

km4ack, 2021-01-14:
Try reinstalling using the Build a Pi link found in the preferences sub section of the main pi menu. I installed both FLRIG & FLDIGI successfully this afternoon using Buster 02DEC2020 on a Pi 3.

vbursucianu, 2021-01-15:
Just tried it and it still fails. Will send you my make file output but i have to relaunch the Build a pi

vbursucianu, 2021-01-15:
Here is the output of the new run:
```
make[1]: Leaving directory '/home/pi/Downloads/fldigi-4.1.17/po'
Making install in doc
make[1]: Entering directory '/home/pi/Downloads/fldigi-4.1.17/doc'
make[2]: Entering directory '/home/pi/Downloads/fldigi-4.1.17/doc'
make[2]: Nothing to be done for 'install-exec-am'.
 /usr/bin/mkdir -p '/usr/local/share/man/man1'
 /usr/bin/install -c -m 644 fldigi.1 flarq.1 '/usr/local/share/man/man1'
make[2]: Leaving directory '/home/pi/Downloads/fldigi-4.1.17/doc'
make[1]: Leaving directory '/home/pi/Downloads/fldigi-4.1.17/doc'
Making install in src
make[1]: Entering directory '/home/pi/Downloads/fldigi-4.1.17/src'
make  install-recursive
make[2]: Entering directory '/home/pi/Downloads/fldigi-4.1.17/src'
make[3]: Entering directory '/home/pi/Downloads/fldigi-4.1.17/src'
  CXXLD    fldigi
/usr/bin/ld: dialogs/fldigi-confdialog.o: file not recognized: file truncated
collect2: error: ld returned 1 exit status
make[3]: *** [Makefile:3154: fldigi] Error 1
make[3]: Leaving directory '/home/pi/Downloads/fldigi-4.1.17/src'
make[2]: *** [Makefile:8643: install-recursive] Error 1
make[2]: Leaving directory '/home/pi/Downloads/fldigi-4.1.17/src'
make[1]: *** [Makefile:8974: install] Error 2
make[1]: Leaving directory '/home/pi/Downloads/fldigi-4.1.17/src'
make: *** [Makefile:469: install-recursive] Error 1
Resetting swap size to default
[ ok ] Restarting dphys-swapfile (via systemctl): dphys-swapfile.service.
```
I am not sure but maybe the default 100MB Swap is too little and needs to be higher and could explain the crash....will try that

km4ack, 2021-01-15:
On a Pi 3 the 100MB swap is an issue. However, the script should automatically increase to 1024 before the build and then reset it to 100 after the FLDIGI build is complete. Hence the line "Resetting swap size to default"

vbursucianu, 2021-01-15:
I've manually increased the swap size to 1Gb and after running the build a pi installation for FLDIGI, it failed again And it decreased my swap size back to 100Mb!

km4ack, 2021-02-17:
Did you get this resolved? If not, you might want to try the [3.0.6 beta](https://groups.io/g/KM4ACK-Pi/topic/build_a_pi_3_0_6_beta/80640680?p=,,,20,0,0,0::recentpostdate%2Fsticky,,,20,2,0,80640680)

km4ack, 2021-07-05:
Work is currently in process to update the FLDIGI install as a new library is required for successful installation. See https://github.com/km4ack/pi-build/issues/263

closed 2021-07-05 by km4ack