km4ack/pi-build #204: Error when installing using the git clone code.

KB2RLB, 2021-01-18:
After multiple clean OS installs, I receive the following error after adding my callsign.
Using Pi 3b+ and latest Buster OS.

Processing triggers for install-info (6.5.0.dfsg.1-4+b1) ...
Processing triggers for man-db (2.8.5-2) ...
/home/pi/pi-build/build-a-pi: line 100: /home/pi/pi-build/intro.txt: Permission denied
/home/pi/pi-build/build-a-pi: line 112: /home/pi/pi-build/intro.txt: No such file or directory
rm: cannot remove '/home/pi/pi-build/intro.txt': No such file or directory
/home/pi/pi-build/build-a-pi: line 169: /home/pi/pi-build/config: Permission denied
/home/pi/pi-build/build-a-pi: line 174: /home/pi/pi-build/base.txt: Permission denied

----------
Beginner level with Linux/Pi coding (but not dumb as a rock)..lol

Thanks
Tim

km4ack, 2021-01-21:
Hi Tim,

I think you posted this issue to the [forum](https://groups.io/g/KM4ACK-Pi/topics) as well. Did you get it resolved?

73, de KM4ACK

KB2RLB, 2021-01-21:
Jason,
Yes I did get it resolved. I had to download another copy of Buster (the
one I had seemed to be corrupt).

Thanks,
73, AC1KD

closed 2021-01-21 by km4ack