km4ack/pi-build #210: conky gps not working

bulldozer1982, 2021-02-02:
I ran into an issue with my gps not working.  I performed the fix from the youtube video and everything was working find.  I then shutdown pi and the next day turned it on and my gps was working.  I ran the cgps command and all data is showing n/a except status which is saying no fix.  can anyone help me fix this issue so i can have the gps up and running.  thanks for the help.

KE0OOS, 2021-02-06:
I'm having a similar problem on my 7 inch .. which is running the 3.0.6 beta.

km4ack, 2021-02-06:
Are you positive that the GPS has a lock on the satellites? Which GPS are you using?

KF5KUW, 2021-02-07:
I have found that CGPS and XGPS both return with no data, but if I run the GPS Update Tool, this detects the GPS, and after it finishes both CGPS and XGPS are showing satellite results.
I am testing this with the beta 3.0.6.
Also, my GPS stays powered up (plugged into powered USB hub connected to RPi4b), and I am using a Onyehn VK-162 G-Mouse USB GPS Dongle Navigation Module.

KE0OOS, 2021-02-13:
My situation is simular to KF5KUW in his comment.   I also found by restarting gpsd using:
sudo systemctl restart gpsd
will get the GPS display working properly, and also CGPS, XGPS.

km4ack, 2021-03-15:
Install the latest BAP 3.0.7 and then use BAP to re install Conky. I can reopen this issue ticket if the problem persists after taking these steps.

closed 2021-03-15 by km4ack