km4ack/pi-build #213: Conky redraw incorrect on boot

km4ack, 2021-02-07:
Conky is double drawing on boot with a pi 3 (Buster 02DEC2020) which make it completely unreadable and unusable. I have found the easy fix is to open the Pi Menu. Under the ham radio section, click Conky. This forces Conky to restart and the issue won't return until the next boot. Need to figure out what is causing the issue on boot.

KF5KUW, 2021-02-07:
Problem also exists with a Pi 4b,w/4Gb RAM.

KE0OOS, 2021-02-13:
I'm noticing the problem when I power up the BAP.   I'll test further if the problem is on reboot, but it is looking "normal" when I did a reboot this morning.

Running P4b, 4Gb Ram on a 7 inch Pi screen.

km4ack, 2021-03-05:
as a workaround for the time being, I add this line to crontab
@reboot sleep 30 && export DISPLAY=:0 && /usr/bin/conky

which will restart conky clearing the problem

km4ack, 2021-04-10:
possible solution on this page - https://askubuntu.com/questions/567737/why-does-conky-superpose-system-data-when-updating-its-output

km4ack, 2021-07-11:
See [this](https://groups.io/g/KM4ACK-Pi/message/4650) for possible resolution to this issue.

commit fec67676 (2021-07-12): fix #213
closed 2021-08-05 by km4ack