km4ack/pi-build #219: build a pi

ws9b, 2021-02-25:
I am using a mfj-1204 interface.  I had Pat, yaac and xastir working.  Then I installed build a pi.  Now my radio does not transmit for APRS or PAT and I cannot receive APRS also. 
I also cannot get menu items boxes small enough to see buttons at bottom.  I am using a pi 3B+ and 7" monitor.
any help would be appreciated.
Thanks

ws9b, 2021-03-02:
am update to my problems with build a PI.
I get the following message when I start PAT from command line:
Unable to get frequency from rig my-rig: dial tcp 127.0.0.1:4532: connect: connection refused.
I do not have rig control load and I am not using it.  I did remove rig control lline form PAT menu settings, but did not help.
Thanks
Tom

closed 2021-03-02 by ws9b

km4ack, 2021-03-02:
Hi Tom,

This seems more of a configuration issue instead of a bug. Post the question to the forum and let's see if we can get you up and running. https://groups.io/g/KM4ACK-Pi/topics

73, de KM4ACK