km4ack/pi-build #223: Gridtracker not updating

km4ack, 2021-03-02:
not updating to the latest version. see [this post](https://groups.io/g/KM4ACK-Pi/topic/81015410)

chrisgaeth, 2021-04-02:
Jason,
To fix this you just need to update your Gridtracker GRIDDLL variable to `GRIDDLL=$(curl -s https://gridtracker.org/downloads/ | grep arm | head -1 | sed 's/.*href="//' | sed 's/.gz.*$/.gz/')

PA2EJD, 2021-05-17:
Jason,
The actual URL for the current version is https://fleetwood.mchambersradio.com/gridtracker/v1.21.0407/GridTracker-1.21.0407-linux-arm.tar.gz Your script resolves the correct tar file name using the url given by chrisgaeth but does not generate the correct url.

I was able to install Gridtracker manually following your lines of code with the new url.

Thanks for the great work on Build a Pi

lcgreenwald, 2021-06-08:
Jason,
This cleans up the HTML formatting and generates the correct URL:
GRIDDLL=$(curl -s https://gridtracker.org/downloads/ | grep arm | head -1 | sed 's/<p style="padding-left: 80px;"><a href="//' | sed 's/.>GridTracker.*//')

73
Larry - wb0sio

lcgreenwald, 2021-06-08:
Not sure what corrupted my previous post but I'll try the post again. 
`GRIDDLL=$(curl -s https://gridtracker.org/downloads/ | grep arm | head -1 | sed '/.*href="//' | sed 's/.>GridTracker.*//')

73
Larry - wb0sio

dave2492, 2021-06-17:
Thanks, Larry.  But I got an error - missing the first sed command.  So, would be:
GRIDDLL=$(curl -s https://gridtracker.org/downloads/ | grep arm | head -1 | sed 's/.*href="//' | sed 's/.>GridTracker.*//')
But then got another error: ERROR 503: Service Temporarily Unavailable.
It appears that cloudflare is blocking access unless you are using a browser (DDOS protection).
See: https://superuser.com/questions/888507/problems-with-wget-to-a-cloudflare-hosted-site-503-service-unavailable
I ended up downloading with the browser and untar'd where build-a-pi expected gridtracker to live.
Dave.

km4ack, 2021-07-05:
a resolution is in the dev brach of code and will be included in 3.0.8 https://github.com/km4ack/pi-build/commit/e95918681bab5856a7812e450beb5dfc2ae44074

km4ack, 2021-08-05:
Fixed in 3.0.8 release

closed 2021-08-05 by km4ack