km4ack/pi-build #225: qsstv 9.5.1 not installing latest version

km4ack, 2021-03-02:
i haven't successfully gotten the latest version of qsstv (9.5.1 02MARCH2021) to build on a Pi. As such, the BAP code was changed and hard coded to install 9.4.4 until this can be resolved. Per the [qsstv website](http://users.telenet.be/on4qz/qsstv/) 9.4.x is still the latest official version as of 02MARCH2021. 

_Screenshot of website 02MARCH2021_

[image]

falmx, 2021-06-29:
Did a build on a fresh installed Pi3, no issues. Is there a specific error you get, which Pi are you using. A Pi4?
[image]

km4ack, 2021-07-06:
I am running on a Pi 3 as my test box. When I run "make", I get the following:

g++ -c -pipe -O3 -Wno-implicit-fallthrough -Wno-psabi -std=gnu++11 -Wall -W -D_REENTRANT -fPIC -DQT_DEPRECATED_WARNINGS -DQT_NO_DEBUG -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_XML_LIB -DQT_CORE_LIB -I. -Iconfig -Idispatch -Idrmrx -Idrmtx -Idsp -Ieditor -Ilogbook -Imainwidgets -Irig -Iscope -Isound -Isstv -Iutils -Iwidgets -Ixmlrpc -Ivideocapt -isystem /usr/include/openjpeg-2.3 -isystem /usr/include/arm-linux-gnueabihf/qt5 -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWidgets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtGui -isystem /usr/include/arm-linux-gnueabihf/qt5/QtNetwork -isystem /usr/include/arm-linux-gnueabihf/qt5/QtXml -isystem /usr/include/arm-linux-gnueabihf/qt5/QtCore -I. -I. -I/usr/lib/arm-linux-gnueabihf/qt5/mkspecs/linux-g++ -o rigcontrol.o rig/rigcontrol.cpp
rig/rigcontrol.cpp: In member function ‘bool rigControl::init()’:
rig/rigcontrol.cpp:90:99: error: ‘FILPATHLEN’ was not declared in this scope
    strncpy(my_rig->state.rigport.pathname,(const char *)catParams.serialPort.toLatin1().data(),FILPATHLEN-1);
                                                                                                ^~~~~~~~~~

make: *** [Makefile:5030: rigcontrol.o] Error 1

km4ack, 2021-07-06:
I should add that I have hamlib 4.2 installed already.

patrickdickey52761, 2021-07-22:
I was able to install QSSTV 9.5.6 today before checking out the beta version of Build-A-Pi. I went through their FAQ page and made sure that I had all of the dependencies (which we do) and checked the QMake version (it was higher than what their FAQ shows). I followed their instructions (download the tar.gz file, unzip and change to qsstv directory, qmake, make, sudo make install) and it worked. This is on a Raspberry Pi 4.

km4ack, 2021-07-23:
Thanks for the report @patrickdickey52761 It was successful on my Pi 4 as well but still won't build on the Pi 3. Not sure what the difference is. I'll have to figure out how I am going to handle that in BAP.

km4ack, 2021-07-24:
I think I finally got to the bottom of this and have it installing on both the Pi 3 & 4. Pretty sure it was user error on my part. Code has been updated and pushed to beta 3.0.8.

km4ack, 2021-08-05:
Fixed in 3.0.8 release

closed 2021-08-05 by km4ack