km4ack/pi-build #229: FLrig fails to run in BAP 3.0.6 Beta

JaySheepdog-N0SR, 2021-03-10:
Using the latest RaspberryPi OS (11 Jan 2021) (with desktop and recommended software) and BAP 3.0.6 beta. 

After uneventful install, discovered that flrig requires libjpeg.so.9 module, which is part of the libjpeg9 package. This package was not installed, although libjpeg-dev was current. 
The fix was a simple:

**sudo apt install libjpeg9**

I did not install all software in BAP (although all software on the system was installed by BAP), the following were NOT installed:
- CHIRP
- GARIM
- M0IAX
- XLOG
- QSSTV
- HAMCLOCK
- DIPOLE
- GPARTED
- RTC
- TELNET
- SECURITY
- YGATE
- ZYGRIB

One of the above packages may install libjpeg9. 
One interesting thing I didn’t realize until I scanned for installed packages was that M0IAX and XLOG were not installed, although I could have missed the check boxes when running BAP initially.

km4ack, 2021-03-10:
Thanks for the report. Downloading the latest OS now and will test. XLOG will not currently install. See this [issue](https://github.com/km4ack/pi-build/issues/215) for details. Was your install on a Pi 3 or 4?

km4ack, 2021-03-10:
Just finished a new build but only installed FLRIG. Pi 3/RaspberryPi OS (11 Jan 2021) Desktop only software/BAP 3.0.6. FLRIG installed and ran ok. I will start another build and install the same items you list above to see if I can reproduce the issue.

JaySheepdog-N0SR, 2021-03-10:
Reproducible is fixable!
My instinct tells me that after flrig (and associated programs) are installed that another package down the line installs something from the repo that tells apt that libjpeg9 is no longer needed and removes it.

km4ack, 2021-03-10:
Exactly! I am running another build now and something (not sure what) has said that libjpeg9 is no longer needed. Line 564 in [build-a-pi](https://github.com/km4ack/pi-build/blob/master/build-a-pi) is an autoremove command.

closed 2021-03-10 by km4ack

JaySheepdog-N0SR, 2021-03-10:
Well what do you know. A broken watch is correct twice a day! Lol. I’m sure you’ll be able to track it down fairly quickly. 
73
Jay

km4ack, 2021-03-11:
for now i simply commented out the autoremove command.

commit 64c088f6 (2021-03-10): fix #229