km4ack/pi-build #238: Error message when trying to run Build-a-Pi

tarjeibr, 2021-03-20:
[image]
Installasjon went well, but afterwards when I run «build-a-pi» and start a scan I unfortunately had disconnected the nettwork cable and I got this error message. After I Connect nettwork and I verifiser that internett worked I still get the same error, Even after reboots. I am a novise with Pi and need help about What to do.

2I0YUM, 2021-03-23:
I'm getting the same error too fresh OS build.

km4ack, 2021-03-24:
reference this [post](https://groups.io/g/KM4ACK-Pi/topic/error_when_running/81497591?p=,,,20,0,0,0::recentpostdate%2Fsticky,,,20,2,0,81497591)

closed 2021-03-24 by km4ack