km4ack/pi-build #250: Delete Alias not working right

KD7UM, 2021-04-08:
When I used Pat Menu >> Manage PAT >> Delete Alias  none of the alias listed are my Pat alias.  I am not sure where this function is pulling the list from. 

It only shows ARDOP listings of which none are mine

km4ack, 2021-04-08:
In the Pat Mailbox, click on Action->Connect. The top left box has a dropdown to show current aliases. This is where the "delete alias" list pulls from and allows you to delete. I am running 2.3.0 and just verified that it is working correctly on my system. You may need to refresh the Pat Mailbox before additions/deletions show correctly.

km4ack, 2021-04-08:
This is a pat menu issue instead of BAP. New issue ticket created in pat menu repository. https://github.com/km4ack/patmenu2/issues/26

closed 2021-04-08 by km4ack