km4ack/pi-build #251: IC-7300 and Pat Winlink - Ardop

AC1KO, 2021-04-11:
Has anyone gotten Pat Winlink working under build-a-pi?  I just got a brand new pi 4-8G, did a fresh buster image and installed build-a-pi (all).  I have wsjt-x working with FT-8 and wspr, FLrig and FLdigi working and for FLrig-all controls work on the radio.  However, with Pat winlink and Ardop... will not key the radio, or will not stop keying the radio depending on if I have the daemon for hamlib running or not.   Do I really need hamlib considering I can change modes etc without rigctld running, but PTT isn't coming across.  When I run the daemon: 
rigctld -m 3073 -r /dev/ttyUSB0 -s 115200
I can do winlink with ardop on the windows machine using winlink express... has anyone gotten pat winlink to work on a pi using ardop with a 7300 (or any modern radio with a built-in sound card?) thanks

km4ack, 2021-04-12:
Many including myself have pat winlink running on the pi. This isn't a BAP issue per say. Please post the question in the forum for help. https://groups.io/g/KM4ACK-Pi/topics Also, while you are there, check out the FAQ pinned post where other 7300 users have submitted setup guides.

closed 2021-04-12 by km4ack

AC1KO, 2021-04-12:
Thanks Jason, and keep up the good work on the videos.. I've dabbled with
Linux in the past and never went far with it other than to get a bit more
life out of an old computer, eliminating Microsoft bloatware, but you
inspired me to try out the raspberry pi...  you and OH8STN and by the way,
I watched your April 1 video before I had my coffee and I had just ordered
my first raspberry pi the day before... you got me good! :P

Are you thinking about doing a step-by-step setup video for the 7300?
Seems like a lot of them out there.  My goal is to eventually get a go box
together for field day and/or SHTF with a pi and a 705 where I could see a
real use for winlink.  I think my issue comes down to PTT function.. the
FLRIG PTT button keys the radio and unkeys when I toggle... Ardop doesn't
key it at all, and if I run the Hamlib script it keyes and never lets go
until I reboot the pi.

thanks
Matt

73 de AC1KO

AC1KO, 2021-04-12:
I didn't have to post I read some of the info on the forum and got it
working!!!!  Not running the daemon either... just confirmed I had the same
settings and in FLRIG, changed from using RTS to CAT for PTT and viola!  It
works!  Backing up the image as we speak.

73 de AC1KO

km4ack, 2021-04-13:
Good deal. Glad you got it up and running :-)