km4ack/pi-build #253: GridTracker upgrade neded

MikeWills, 2021-04-21:
Maybe it's related to the bug I found, but GridTracker has updated. Wil the upgrade script prompt me to update it or how do I go about upgrading it "properly"?

PD7RON, 2021-05-04:
Yes I have the same problem.
Gridtracker says their is an update. In build-a-pi there is no update needed. What is the way to get it updated.

MikeWills, 2021-05-04:
#223 I think is the fix. But either I don't know what I'm doing or I edited the wrong spot to test it.

darigovresearch, 2021-05-30:
@MikeWills & @PD7RON there has been an additional post on the issue https://github.com/km4ack/pi-build/issues/223 which may help to resolve your problem

Could you have a go to see if this is the case?

MikeWills, 2021-05-31:
I made that change but still doesn't work. If you look at the latest, it's not generating the right URL to detect the latest version. So it's not a fix and I don't have the time nor the bash skills to try to troubleshoot it right now.

PD7RON, 2021-08-08:
> Closed #253 via e959186.

Hello Jason
How and where do I have to use this change in the software for gridtracker.
The update for Girdtracker  is not working. And also the update for flamp and flmsg keeps asking for a needed update.
Further-on I can’t figure-out how to start linbpq. The line ~/linbpq/.linbpq  is not working for me

73 Ron PD7RON

MikeWills, 2021-08-09:
Same issues, GridTracker is not showing an update for me.

km4ack, 2021-08-09:
have you updated to BAP 3.0.8?

MikeWills, 2021-08-09:
Yes, I have. Updated, installed some updates for FLDIGI package, rebooted 3 more times. Nothing.

MikeWills, 2021-08-09:
Attached some screenshots. 
[image]
[image]

MikeWills, 2021-08-09:
One more of the screen. 
[image]

km4ack, 2021-08-09:
Thanks for the report. Reopening this issue. Will try to figure out what the bug is.

MikeWills, 2021-08-09:
If it helps, I'm on a 4GB Pi4. It shouldn't matter, but I know computers are weird. Let me know if you want some more testing. I am willing to help out if you want to try some things.

km4ack, 2021-08-09:
Even though BAP says the latest version is installed, what happens if you put a checkmark by it and try to install it anyway? I just installed gridtracker on my test box and 1.21.0620 did install. I did not have a previous version on that pi.

MikeWills, 2021-08-09:
Okay, there appears to be a script error. I got this during the install.  After rebooting, 1.20 is still installed. 
[image]

PD7RON, 2021-08-09:
Hello Mike and Jason
Yes I also have 3.0.8 installed. And Mike I got also a message that the installed version of Gridtracker is still 1.20.
For Flmsg and flamp there is also a message that there is an error.
I am using a raspberry pi 4b with 4G
Testing is ok for me let me know.

73 Ron 
PD7RON

PD7RON, 2021-08-09:
I forgot to tell that I put a checkmark to install Gridtracker again. But it is still the 1.20 version

73 Ron

km4ack, 2021-08-09:
can one of you guys run the following command:
`cd ~ && ls`
and post the output. I think I have spotted the issue.

MikeWills, 2021-08-09:
[image]

km4ack, 2021-08-09:
Thanks Mike. First, [backup your pi](https://youtu.be/OcL7Ul-ajz0)....just in case. Next we want to remove the old gridtacker files with
`rm -rf ~/GridTracker`
finally, we want to move the new gridtracker directory to the correct spot.
`mv ~/GridTracker-1.21.0620-linux-arm ~/GridTracker/`
Then open GridTracker from the Pi menu and verify the correct (1.21.0620) is showing as expected.

MikeWills, 2021-08-09:
That did it! Up and running.

km4ack, 2021-08-09:
Thanks for confirming. I am going to open another issue and close this one since this problem is a bit different.

closed 2021-08-09 by km4ack

PD7RON, 2021-08-09:
Yes for me also up and running.
Thanks 

73 Ron

PD7RON, 2021-08-09:
Jason can you tell me how I start linbpq. The line ~/linbpq/.linbpq is not working for me.

73 Ron
PD7RON

km4ack, 2021-08-09:
Thanks for the report Ron. The file placed on the desktop gives the wrong command. The correct command is
`~/linbpq/./linbpq`
I'll file a new issue ticket so this can be addressed in the future.

PD7RON, 2021-08-09:
Ok tnx Jason.
I had  already tried 
cd linbpq 
./ linbpq
And saw that it was working.

73 PD7RON
Ron

commit e9591868 (2021-07-05): fix #253