km4ack/pi-build #256: Add piQtSoundModem and piQtTermTCP

5rm, 2021-05-08:
As you added BPQ (pilinbpq) those two are important additions. QtSoundModem is better AX25 packet (and more) modem for HF than DW.
https://www.cantab.net/users/john.wiseman/Downloads/Beta/piQtSoundModem
piQtTermTCP is a terminal with TCP and AWG interfaces to work with BPQ and Soundmodem
https://www.cantab.net/users/john.wiseman/Downloads/Beta/piQtTermTCP

Jouko OH5RM

km4ack, 2021-07-06:
piQtTermTCP was added in version 3.0.5. It is available on the last screen (utilities) during the build. Will consider QtSoundModem. Thanks for the suggestion!

km4ack, 2021-07-06:
Can you point me to the documentation for QtSoundModem?

5rm, 2021-07-06:
Thanks Jason,
PiQtSoundmodem is much handier than dw asit has a gui to set things up and also waterfall which is almost a must on hf. It is a linux port of the famous uz7ho soundmodem.
73
Jouko oh5rm

5rm, 2021-07-06:
https://www.cantab.net/users/john.wiseman/Documents/QtSoundModem.html

Terveisin
Jouko Nurma

Darrrren, 2021-07-10:
This will "level up" the build considerably

km4ack, 2021-08-05:
Included in 3.0.8 release

closed 2021-08-05 by km4ack