km4ack/pi-build #257: No Permission when entering the "Call Sign"

kauffmanlll, 2021-05-10:
I opened up the terminal, I validated that I had the default passwd set for pi and I pasted the link and it began to pull the files. It opend the KM4ACK graphical install and prompted for the call sign. I submitted 'KE0BZW' which is valid with ARRL and the program spit out that it did not have permission, followed by exiting.

km4ack, 2021-07-07:
Was this on a fresh install of Buster? Is your username still pi?

KJ7RUP, 2021-07-13:
I'm running into this same issue.  I've confirmed my username is pi and that the password is the default "raspberry".  This is the first thing I'm trying to do after a fresh install + updates of the image you linked to in your instructions.

When running the clone command, here are the errors I get:
Line 100: /home/pi/pi-build/intro.txt: Permission Denied
Lone : /home/pi/pi-build/intro.txt: No such file or directory
rm: cannot remove '/home/pi/pi-build/intro.txt': No such file or Directory

At this point, the box for entering my call sign pops up.  I enter my call sign and...

Line 174: /home/pi/pi-build/base.txt: Permission denied

At this point the process stops and I'm back to the beginning in the terminal, but have to delete the pi-build folders before I can start the process again, otherwise I get errors saying that the folders already exist and are not empty.

km4ack, 2021-07-13:
> I'm running into this same issue. I've confirmed my username is pi and that the password is the default "raspberry". This is the first thing I'm trying to do after a fresh install + updates of the [...]

It looks like you may not have been in the home directory when you ran the script. Make sure you are in your home directory by running `cd` and pressing enter. Then run the ` git clone https://github.com/km4ack/pi-build.git \
  $HOME/pi-build && bash $HOME/pi-build/build-a-pi` Also, be sure not to use sudo for any of the commands.

KJ7RUP, 2021-07-21:
That did it.  I'd been trying to run the lines one at a time and using sudo- (total noob at linux and didn't watch the video all the way through! It worked great.  Thanks for all the work!

closed 2021-07-21 by km4ack