km4ack/pi-build #260: FT dx10 not recognized by flrig 1.3.54 in Build-a-Pi

KG4IIC, 2021-06-09:
I run R Pi 4B 2GB and CAT control the FT-dx10 via flrig: confi > xcvr > rig.  FT dx10 is not listed and instead I select FT-dx101D. This works with FLRig FLRig.  WSJT-X and CQRLog work with FLRig FLRig selected.  
WSJT-X 2.3.1 has rig: FT-dx10 as a selection.

km4ack, 2021-07-05:
This isn't a Build a Pi issue but a FLRIG issue. Check the FLRIG forum for solution https://groups.io/g/linuxham/topics

closed 2021-07-05 by km4ack

KG4IIC, 2021-07-07:
Thanks.  I pursue this with the FLRIG forum.  Thanks for all the good work. 73

BTW  Elmore County ARC had a great field day roughing it on Bennet Mountain north of Mountain Home, ID.  Just a small group of us having fun making QSOs.