km4ack/pi-build #261: Grid Tracker Will Not Download

NZ1N, 2021-06-21:
I had an issue with an update and reflashed my SD with buster. I went and redid Build-a-Pi and everything seems to be working fine except Grid Tracker. I get an error that says "Invalid desktop entry file:' /usr/share/applications/gridtracker.desktop' I have redone everything from the beginning now twice. Same issue. Am I missing something?
Note: I am a watch a video and follow step by step kind of guy so if someone know a fix please pretend I'm 9 years old when explaining the fix.
Thanks,
Will 
NZ1N

KN4USA, 2021-06-22:
My issue is similar. I did a fresh install and a number of items will not install. Xlog, QSSTV, Gridtracker are all showing "Not Installed"  when I run updater.

snwvo7, 2021-07-02:
I have exactly the same problem  Xlog, QSSTV, Gridtracker are all showing "Not Installed" when I run updater. and won't install after the second try

km4ack, 2021-07-05:
Duplicate issue already reported https://github.com/km4ack/pi-build/issues/253 Hope to resolve in 3.0.8.

closed 2021-07-05 by km4ack