km4ack/pi-build #262: 2 programs are not installed and Conky problem

DC2JR, 2021-06-23:
in 3.0.7 
Gridtracker and QSSTV are not installed (now iinstalled manually)
and Conky information is too high, so that the Callsign disappears under the menubar

km4ack, 2021-06-24:
gridtracker not installing is a known issue https://github.com/km4ack/pi-build/issues/223

You can use Conky Prefs found in the main pi menu under Preferences to resize conky for different screen resolutions. 

I'll take a look at QSSTV and see if I can replicate the issue. 

73, de KM4ACK

DC2JR, 2021-06-24:
Thanks for your quick response. A short hint regarding ptt problems with for example fldigi: user pi is not member of group dialout, after adding pi to dialout, ptt functions

By 73 de Dc2jr Klau

DC2JR, 2021-06-25:
Hello, today when running build-a-pi, I sav that there was an update for fldigi to Version 4.1.19 availlable. I started the update process. The compile run stopped with following error: fatal error: libudev.h: Datei oder Verzeichnis nicht gefunden (file or directory not found) vy 73 de Klaus, DC2JR Gesendet von Mail für Windows 10 Von: KM4ACKGesendet: Donnerstag, 24. Juni 2021 11:54An: km4ack/pi-buildCc: DC2JR; AuthorBetreff: Re: [km4ack/pi-build] 2 programs are not installed and Conky problem (#262) gridtracker not installing is a known issue #223You can use Conky Prefs found in the main pi menu under Preferences to resize conky for different screen resolutions.I'll take a look at QSSTV and see if I can replicate the issue.73, de KM4ACK

closed 2021-07-05 by km4ack