km4ack/pi-build #266: FLDIGI 4.1.19 gpio not working

arvdsar, 2021-07-13:
Hi,

fldigi 4.1.19 (latest version) has broken the GPIO interface for PTT with the Raspberry Pi. 
After downgrading 4.1.19 to 4.1.18 manually, the PTT GPIO is working.

To make sure it was fldigi and not the image, I have taken another image with 4.1.18 which worked. After upgrading to 4.1.19 the GPIO PTT failed again. 

I haven't been able to provide feedback to the creators of fldigi but you might want to stick to 4.1.18 for now instead of including 4.1.19 or maybe want to put a remark in your release notes.

keep up the good work!!

Best regards,
Alexander

km4ack, 2021-07-13:
Thanks for the heads up. You can reach the FLDIGI developer in this forum https://groups.io/g/linuxham/topics

73, de KM4ACK

km4ack, 2021-08-05:
As this isn't a BAP issue, I am closing this issue and leaving the documentation tag so it can be referenced if needed in the future.

closed 2021-08-05 by km4ack

arvdsar, 2021-08-08:
The issue seems to be solved in FLDigi 4.1.20 (mind the 'enable GPIO PTT' and Initialize button in the GPIO config screen)