km4ack/pi-build #277: FLDigi wont install

Quasi2021, 2021-07-31:
Jason, Good afternoon. First I want to thank you for all the work you did on this. I have been stuck in a holding pattern for years trying to install a different OS on my RPi's from a MacBook... I spent last night and today installing and building my RPi3 and mostly everything worked, after doing it twice. lol. Anyway, for some reason FLDgigi will not load on my Pi. Hopefully this picture shows up... Thank you in advance!
[image]

km4ack, 2021-07-31:
it looks like you are having some internet connection issues since it shows you can't resolve raspberrypi.org. There is also an issue in 3.0.7 with FLDIGI/FLRIG which has been [documented](https://groups.io/g/KM4ACK-Pi/message/4538).

closed 2021-07-31 by km4ack

Quasi2021, 2021-07-31:
Thank you,
I don’t usually have connection issues with my internet but its possible. I am running the script after entering this command libudev-dev. What does that command actually do? Out of curiosity.
Thank you again in advance