km4ack/pi-build #296: hamlib 4.3 compile error

km4ack, 2021-09-01:
It appears with the new 4.3 release of hamlib, it has a dependency that
the raspberry OS does not have yet.  It needs libusb-1.0.23, but the OS
only has 1.0.22 in the repository.  When you first compile it, it errors
out complaining libusb-1.0 is not installed:

testlibusb.c:30:10: fatal error: libusb-1.0/libusb.h: No such file or
directory

When you install libusb-1.0 (sudo apt-get install libusb-1.0), it
installs 1.0.22.  When you try and compile it again, it fails because
hamlib-4.3 requires libusb-1.0.23 or greater.  This unfortunately causes
rigctld and some components of BAP to not work correctly.

A workaround for now is to install hamlib-4.2 until raspberry OS catches
up with libusb. 

Just wanted to let you know so you can make a change to BAP forcing
hamlib 4.2 to install instead of 4.3 which was just released.

Good luck!

Rich / KB3YBE

km4ack, 2021-09-05:
It looks like we will need to compile the latest version of libusb before attempting hamlib 4.3. This should be a quick fix with a patch for now but needs to be added to the BAP hamlib section of code. Testing is underway.
```
cd $HOME/Downloads
wget https://sourceforge.net/projects/libusb/files/libusb-1.0/libusb-1.0.24/libusb-1.0.24.tar.bz2
tar -xf libusb-1.0.24.tar.bz2
cd libusb-1.0.24
./configure
make
sudo make install
```

km4ack, 2021-09-06:
[patch](https://github.com/km4ack/pi-build/blob/dev/patch/hamlib4.3) added 20210905

JohnRudolfLewis, 2021-09-09:
After applying this patch, I'm unable to use either WSJT-X or JS8Call using hamlib rigctld.

Error from JS8Call: Rig Failure. Hamlib error: Communication timed out while getting current VFO frequency.

Error from WSJT-X: Rig Failure. Hamlib error: Communication timed out......<snip>.....git_get_vfo return(-5) while testing getting current VFO.

Both apps worked before this patch and still work when selecting the appropriate rig, but I would prefer to use hamlib.

I'm still troubleshooting, and will post back here if I encounter a positive outcome.

km4ack, 2021-09-09:
John,

Hopefully you backed up the system before applying the hamlib 4.3 patch. I ran into a [similar issue](https://github.com/Hamlib/Hamlib/issues/789) when I upgraded my system from 4.2 to 4.3. I was hoping this was an isolated incident and was because of other things I was doing on the test box. Please keep me posted. I have directions [here](https://groups.io/g/KM4ACK-Pi/message/5509) for manually building 4.2.

km4ack, 2021-09-09:
I have pulled the patch for hamlib 4.3 until more testing can be done to confirm there isn't a widespread issue when upgrading from 4.2 to 4.3. The original 4.2 patch has been restored for new Build a Pi builds which will not be able to compile 4.3 on a new system using BAP.

km4ack, 2021-09-09:
@JohnRudolfLewis Which radio are you using? Just wondering if it is the Icom 705 which is the radio I was using when I had the issue updating hamlib.

JohnRudolfLewis, 2021-09-09:
Right now, I'm using a ubitx which emulates an FT-817.
No I did not backup my system unfortunately.

JohnRudolfLewis, 2021-09-09:
Ok, I'm back up and running...

I had to run the following commands to get it working again:

cd ~/Downloads/hamlib-4.3
sudo make uninstall
cd ~/Downloads/libusb-1.0.24
sudo make uninstall
sudo ldconfig
cd ~/Downloads/hamlib-4.2
./configure
make
sudo make install
sudo ldconfig

Everything is fine now. Reach out if you'd like me to test something related.

JohnRudolfLewis, 2021-09-09:
And my discovery on how to fix my problem plus your assertion that a clean install of 4.3 worked led me down another experiment....

before installing 4.3, do a sudo make uninstall on 4.2.... everything seems to work now. I'd suggest adding that to your patch and re-testing.

km4ack, 2021-09-10:
as of 10SEPT2021 hamlib 4.3 is compiling correctly on a fresh build of BAP.