km4ack/pi-build #304: Clean build on Pi3B and several FL apps are not installing

KD7UM, 2021-09-19:
I did a clean build on a Pi3B and after the build I was missing three FL apps:  FLDigi, FLAmp, and FLMsg.  Subsequent attempts using BAP Update show they are not installed so when I select them, the script runs but at the end  of a period of time they are still not there or are missing from the FLSuite menu.  When I left it running after selecting these apps it looked like it was compiling them.  FLDigi version it is attempting to install is V4.1.20

I did not apply the patch for Hamlib 4.3 before doing this.

I saw this at the end of the FLdigi compile:
-----------------------
fi
make[1]: Leaving directory '/home/pi/Downloads/fldigi-4.1.20/po'
Making install in doc
make[1]: Entering directory '/home/pi/Downloads/fldigi-4.1.20/doc'
make[2]: Entering directory '/home/pi/Downloads/fldigi-4.1.20/doc'
make[2]: Nothing to be done for 'install-exec-am'.
 /usr/bin/mkdir -p '/usr/local/share/man/man1'
 /usr/bin/install -c -m 644 fldigi.1 flarq.1 '/usr/local/share/man/man1'
make[2]: Leaving directory '/home/pi/Downloads/fldigi-4.1.20/doc'
make[1]: Leaving directory '/home/pi/Downloads/fldigi-4.1.20/doc'
**Making install in src
make[1]: Entering directory '/home/pi/Downloads/fldigi-4.1.20/src'
make  install-recursive
make[2]: Entering directory '/home/pi/Downloads/fldigi-4.1.20/src'
make[3]: Entering directory '/home/pi/Downloads/fldigi-4.1.20/src'
  **CXXLD    fldigi
/usr/bin/ld: dialogs/fldigi-confdialog.o: file not recognized: file truncated
collect2: error: ld returned 1 exit status
make[3]: *** [Makefile:3182: fldigi] Error 1
make[3]: Leaving directory '/home/pi/Downloads/fldigi-4.1.20/src'
make[2]: *** [Makefile:8747: install-recursive] Error 1
make[2]: Leaving directory '/home/pi/Downloads/fldigi-4.1.20/src'
make[1]: *** [Makefile:9078: install] Error 2
make[1]: Leaving directory '/home/pi/Downloads/fldigi-4.1.20/src'
make: *** [Makefile:469: install-recursive] Error 1
--2021-09-19 10:06:01--  http://www.w1hkj.com/files/flamp/flamp-2.2.05.tar.gz
Resolving www.w1hkj.com (www.w1hkj.com)... 143.95.246.118
Connecting to www.w1hkj.com (www.w1hkj.com)|143.95.246.118|:80... connected.**
HTTP request sent, awaiting response... 200 OK**
Length: 518387 (506K) [application/x-gzip]
Saving to: ‘flamp-2.2.05.tar.gz’

km4ack, 2021-09-19:
I'll look into the FLDIGI issue as this hasn't been reported to my knowledge. FLAMP is[ an issue](https://github.com/km4ack/pi-build/issues/301) and a patch is available to get it installed using BAP 3.1. A patch is also available for FLMSG. A full description of patches can be [found here](https://github.com/km4ack/pi-build/tree/dev/patch).

km4ack, 2021-09-19:
I just finished a full clean build and FLDIGI 4.1.20 installed on the Pi 3. Can you use the BAP update tool to reinstall FLDIGI and see if it completes please?

KD7UM, 2021-09-20:
Here is what I have done since I posted the issue:

 

1.	Started new clean build on a RPi 3B.
2.	Installed all applications even though I generally skip QSSTV and the YGATE app
3.	Everything installed except FLMSG and FLAMP
4.	Hamlib says it needs an update (4.2?)

 

Went back to the broken system and observed the following:

1.	I did the flamp fix and flmsg patch and they installed but not in the FLSuite folder.  They are in the Internet folder
2.	When I run Build a Pi update it doesn’t find any patches to apply – see below:

##########################################################

#Checking for available patches that haven't been applied#

##########################################################

Initialized empty Git repository in /run/user/1000/pi-build/.git/

Updating origin

remote: Enumerating objects: 1689, done.

remote: Counting objects: 100% (629/629), done.

remote: Compressing objects: 100% (435/435), done.

remote: Total 1689 (delta 384), reused 414 (delta 192), pack-reused 1060

Receiving objects: 100% (1689/1689), 699.86 KiB | 2.05 MiB/s, done.

Resolving deltas: 100% (1046/1046), done.

From https://github.com/km4ack/pi-build

* [new branch]      beta       -> origin/beta

* [new branch]      dev        -> origin/dev

* [new branch]      master     -> origin/master

From https://github.com/km4ack/pi-build

* branch            dev        -> FETCH_HEAD

sh: 0: getcwd() failed: No such file or directory

sh: 0: getcwd() failed: No such file or directory

sh: 0: getcwd() failed: No such file or directory

 

HAMLIB says it is the current version (4.3).  I am not presented with a patch to go back to 4.2.  FLDigi install has failed very time at this point.  So I think my problem was a 4.3/FLdigi issue.  I can try to revert back to Hamlib 4.2 but I need a patch to select from.  Can you help me get the Haml;ib 4.2 patch back and then I will try FLDigi again?

 

This might have been when Hamlib 4.3 was installed by BAP on the initial build?  Now it installs Hamlib 4.2 and everything works OK.

 

Thanks

Dave

km4ack, 2021-09-20:
The patch tool makes a note of each patch that has been applied. We can manually override this. Run `nano ~/.config/patch` and remove the hamlib line. Save and close the file. Now run the BAP update tool and you should see the notice that a patch is available and allow you to apply the 4.2 patch. With that said, I believe there is still an issue around hamlib 4.3 installing correctly. I plan to dig into it this week and see if I can figure it out. The hamlib 4.3 compile issue has been documented [here](https://github.com/km4ack/pi-build/issues/296)

KD7UM, 2021-09-20:
I removed the Hamlib line but the patch tool gave me the standard 4 options, not just Hamlib.

Ran the Hamlib patch and BAP Update still show Hamlib as being current.  Did this patch work?

 

Ran BAP to install FLDigi.  It had the same three makefile errors as before.  FLDigi is not installed.  How can I tell if the patch for Hamlib 4.2 worked?

 

Dave

km4ack, 2021-09-20:
If you only removed the hamlib line from ~/.config/patch, then you should only be able to apply the hamlib patch the second time. The other patches would give you a warning that they were already applied before. I may try to mod this in the future. It was the first release of the patch tool. 
To find out which version of rigctl is installed, you can run
`rigctl --version`
You can also run
`rigctl -vvv -m 1 f`
and see if it produces any errors. FWIW, hamlib didn't install on my test box yesterday yet FLDIGI did install correctly.

KD7UM, 2021-09-20:
I am still running Hamlib 4.3 even though I did the 4.2 patch.

 

***@***.***:~ $ 

***@***.***:~ $ rigctl --version

rigctl Hamlib 4.3 Wed Sep 01 02:14:39 2021 +0000 SHA=5a6134

***@***.***:~ $

 

 

 

Results of rigctl -vvv 1 f

 

 

***@***.***:~ $ rigctl -vvv -m 1 f

Opened rig model 1, 'Dummy'

rigctl_parse: rig not open...trying to reopen

rigctl_parse: RIG_EIO?

main: i/o error

main: rig_close retcode=-1

main: rig_open retcode=0

***@***.***:~ $

 

 

So, it looks like it didn’t uninstall.  My clean build last night installed Hamlib 3.3

 

***@***.***:~ $ rigctl --version

rigctl(d), Hamlib 3.3

 

Copyright (C) 2000-2012 Stephane Fillod

Copyright (C) 2000-2003 Frank Singleton

This is free software; see the source for copying conditions.  There is NO

warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

***@***.***:~ $

 

Let me know how I can help

km4ack, 2021-09-21:
The FLDIGI problem seems to be the result of a corrupted download. The original poster fixed the issue by deleting the FLDIGI folder in ~/Downloads and running the BAP update tool again to install FLDIGI. 

The rig control error was due to [hamlib 4.3 not compiling](https://github.com/km4ack/pi-build/issues/296) correctly. This has been solved for now with a patch and will be corrected in the next release of BAP.

closed 2021-09-21 by km4ack