km4ack/pi-build #310: Remove package and start again

aerofarm, 2021-10-01:
Somewhere in the install, the pi froze up and only a couple apps were installed. Do I need to completely remove build a pi and start again, if so, what is the easiest way to do this.

Is there a way to add programs to the program once the base program is installed?

Nate

km4ack, 2021-10-01:
This is a common misconception. Often the pi will appear to hang in one spot but it isn't really hung. Some applications take a while to install and it can sit on one line long enough that one thinks it has stalled. 

I recommend flashing a new card and starting from scratch.

closed 2021-10-01 by km4ack

aerofarm, 2021-10-01:
ok thanks - thats what I'm doing.  It was stuck in one spot for over 12
hours last night

Nate