km4ack/pi-build #335: Conky Gridsquare INOP under Raspian Bullseye

NullVibes, 2021-11-10:
Conky Gridsquare empty field.

Conky configs point to a _/run/user/1000/gridinfo.txt_ file which exists, but contains only a carrier-return.  (Empty)

NullVibes, 2021-11-10:
The _~/bin/grid_ script handles the Gridsquare entry for Conky, but requires Ruby and associated gems, which were not installed.

To Fix:
_sudo apt install ruby
sudo gem install gpsd_client
sudo gem install maidenhead_

These packages should be added to the main _pi-build_ script(s).

km4ack, 2021-11-10:
Thanks for the report. Not sure what the anomaly is but I have seen it before where these dependencies didn't install. BAP does call for each of them to be installed when the user installs Conky. See https://github.com/km4ack/pi-build/blob/master/functions/additional.function#L143

closed 2021-11-10 by km4ack