km4ack/pi-build #336: Migrate Ruby gpsd to Python3

km4ack, 2021-11-11:
This may be worth investigating. 

see [this post](https://groups.io/g/KM4ACK-Pi/topic/86954136)

btombaugh, 2021-12-01:
i have followed the instruction in Kevin's document in the other post, and updated successfully

btombaugh, 2021-12-01:
This leads to another question -- if that Python script replaces the two Ruby scripts grid and get_grid, which seemed to updated the file /run/user/1000/gridinfo.txt, do we need to disable or replace the process that was performing that update? i.e. is the Ruby script still running in the background? is gridinfo.txt used for anything other than updating the location in Conky?

starbasessd, 2021-12-01:
I did not, not knowing that file was being used for anything. If it is, it's easy enough to add a line to update that file.

starbasessd, 2021-12-01:
What data is in that file? Just gridsquare, or more?

btombaugh, 2021-12-01:
yes, /run/user/1000/gridinfo.txt contains the grid square, to 8 char. Before I had gpsd working, or if the GPS has no 3D coordinate fix, it showed "NO GPS"

starbasessd, 2021-12-01:
What else uses that file, do you know?

km4ack, 2021-12-01:
The gridinfo.txt file is exclusively used by Conky to display the Grid Square in Conky.

starbasessd, 2021-12-01:
Do you need it? Do you want any mods?

btombaugh, 2021-12-01:
so is there a process that can be disabled or removed if we replace with the python script? I see that it's still being updated... Looks like:
#Update GPS maidenhead for conky
*/1 * * * * /home/pi/bin/conky/grid
in crontab

starbasessd, 2021-12-01:
Yes, the .conkyrc file pulls the data from the .py script whenever it runs/updates, rather than force running the ruby grid and get_grid scripts, if you updated the .conkyrc line.

btombaugh, 2021-12-01:
I've commented out the crontab entry, and placed the PyGridsquare.py script into /home/pi/bin/conky. I updated the .conkyrc to call /home/pi/bin/conky/PyGridsquare.py, so it should not have to be invoking the Ruby scripts any longer. Seems to work fine... Thanks for your responses!

km4ack, 2021-12-01:
One word of caution. At one time we called the ruby script direct from Conky to get the gridsquare. This has the potential to cause issues in some cases. It would be better if the python script was run by cron and send the output (grid square) to a simple text file like the ruby script is doing now. BAP uses /run/user/$UID/filename for many of these type operations to reduce read/write cycles to the SD card.

starbasessd, 2021-12-01:
In my humble ( or not ) opinion, running it every minute is major overkill, especially if you save the  location/grid square to a file, too. If you are using a geo-tracker app while in motion, it shouldn't be reading this script anyway. My 6 digit grid square covers a lot of miles both north - south and east - west. If I covered either distance faster than 10 minutes I'd be breaking speed limits...
As to read-write cycles, I think it's a toss up, but there would be a lot more write cycles to 'update' that txt file, which is what actually causes most of the damage (think 'write once, read many' for SDCards) than reading the script, figuring out what the grid square is, and updating the conky display overlay. 
IIRC the issue with calling ruby from conky was it stopped all updates while the ruby scripts were running pausing the screen, etc, it wasn't able to background it. The python script doesn't do it or at least not nearly as bad as ruby did.

km4ack, 2021-12-02:
I can't disagree with you on running it every minute being overkill. The primary reason for this was to get it in conky as quick as possible after boot. A better approach might be braking it into two commands. One that runs at boot and another that updates it every X number of minutes. 

As for the read/write cycles, the dir /run/user/$UID/ is held in RAM and not written to the SD card.

starbasessd, 2021-12-02:
Can be the same script
in crontab:
@reboot <path to>/PyGridsquare.py
\* \* \* \* \* <path to>/PyGridsquare.py \[will run every minute\]
I can give you the modded section in a little bit to write the gridsquare to that txt file.

starbasessd, 2021-12-02:
PyGetgridsquare.py
- - - - -
!/usr/bin/env python3

from gpsdclient import GPSDClient
import maidenhead as mh

client = GPSDClient(host="<host_ip_address>")

for result in client.dict_stream(convert_datetime=True):
     if result["class"] == "TPV":
        lat = result.get("lat", "n/a")
        lon = result.get("lon", "n/a")
        gridsquare = mh.to_maiden(lat, lon)
        print(gridsquare)
        file1 = open("/run/user/1000/gridinfo.txt","w")
        file1.write(str(gridsquare))
        file1.close()
        break
- - - - -
This will save the gridsquare in the file /run/user/1000/gridinfo.txt

starbasessd, 2021-12-02:
In the .conkrc file change
${font Arial:bold:size=18}${color White}Gridsquare ${alignr}${color Yellow}${execi 25 /home/pi/my_get_grid.py | cut -c1-8}
to
${font Arial:bold:size=18}${color White}Gridsquare ${alignr}${color Yellow}${exec cat /run/user/1000/gridinfo.txt}
save and if PyGetgrid.py created the data file, it should display it, instead of the direct output.

btombaugh, 2021-12-02:
that seems to be working. I indented the Python code for the for loop and if statement. The filename for the crontab needs to be PyGetGridsquare.py, which I also had to chmod +x.
It worked after a reboot and came up with the grid right away. I set it for every */5... Thanks again, I think this is a good solution.

starbasessd, 2021-12-02:
Yeah, I hate markdown language (what github uses for this message posting...) it is such a pain to post code (or edit wiki pages when I support motionEye/motionEyeOS), sorry. I could post a modded text script like from the [other site](https://groups.io/g/KM4ACK-Pi/topic/86954136)

starbasessd, 2021-12-02:
Here's the instruction set from the other site modified with the above info:
[PyGetGridsquare.zip](https://github.com/km4ack/pi-build/files/7639097/PyGetGridsquare.zip)

btombaugh, 2021-12-02:
Kevin, the update instructions look good. Two comments:

If the Ruby grid scripts are replaced with the Python script, and the updates made in the crontab to call the Python script instead of the Ruby script, then I don't think that there needs to be any change made in .conkyrc, since it's already going to cat the same file. One should remove or comment the crontab entry for the Ruby script, so that it isn't running both.

Second, you should put your name and contact info in the instructions! Nicely done, sir!

starbasessd, 2021-12-02:
No problem, writing version 1.0 of document, however, it was intended as a set of instructions for Jason to include into Build-A-Pi, and for a few to 'test' to see if it is worth it. 
My personal setup for field includes 3-5 Pis networked, so one would be a host/router/hub/Access Point, and any other machines, tablets, whatever could get GPS and Time (among other services), using a single GPS dongle. I also have a number of PiZeroWs acting as security cameras (with the motionEye/motionEyeOS that I support here on GitHub). 
If anyone feels the need to contact me, they can at KD9EFV at GMAIL dot COM. 
I am an Enterprise (Level 3) IT Help Desk, best known for acting as go between for end users and developer/programmers. Think: Brunette between 2 Blondes. I run into issues that need outside the box solutions.

starbasessd, 2021-12-03:
Building a Pi Buster clean (updated to today) then will install Build-a-Pi to see what the current default installs, then modify my instructions for it, for use by 'current' users' to update/upgrade/install instead of...

starbasessd, 2021-12-03:
@btombaugh @km4ack 
Would it be better to put the PyGetGridsquare.py call in the system crontab, or do you have a need for it in the user (Pi) crontab?

btombaugh, 2021-12-03:
In my opinion, it would be better to put things in shared/common locations, rather than a user's home folder, in the event that anyone creates their own login instead of using "pi" as the only user... So I would vote for the system crontab.

starbasessd, 2021-12-03:
I wasn't referring to the file location (I was taught always use /usr/bin or /usr/sbin for custom apps <sigh>) I was asking which crontab to use to call the program itself. There is a crontab for each user (crontab -e) and a system crontab (sudo crontab -e). If pi is logged in, both the pi user crontab and the system crontab are run. If johndoe is logged in, his crontab and system are run but not pi's. If no one is logged in, only system crontab is run.

btombaugh, 2021-12-03:
right, sorry, I wasn't clear... I would think that best practice would be to put apps and other files into shared locations like /usr/bin, /usr/local, or /opt, depending on the flavor... If we did that, then using the system/root crontab would be better, as it removes any dependency on which user account is used. But because right now all of the files are already in the ~/pi directory, there is already a dependency that would preclude us from removing the user pi anyway, as everything would break...

Since we're working on a pi that fits in the palm of your hand, and not a large, multi-user system, there seems like less need to avoid putting things in the default user's home directory, and that user's crontab. In other words, I think it's safe to assume that 99.9% of users would be logging into their RPi as "pi" so it's fine to keep things as they are.

starbasessd, 2021-12-03:
Interesting. 
In the field, I have multiple users all the time, using email, shared files and documents, KanBan, Whiteboards, etc. 
At home, I use Conky on all my desktop machines, whether Pi, PC, VM, whatever. (Also use neofetch on CLI machines and added a report for Gridsquare \<grin\>). I never 'remove' the Pi user, anyway. Sometimes rename it... I guess it's harder for me to switch back and forth from a Debian type environment, and the RPiOS environment.

starbasessd, 2021-12-03:
Latest instructions for end user implementation. v0.9.9
[PyGetGridsquare.txt](https://github.com/km4ack/pi-build/files/7652455/PyGetGridsquare.txt)

starbasessd, 2021-12-04:
@km4ack BTW, 3.1.3 and Bullseye: They moved vcgencmd (used for temperature monitoring in conky) from /opt/vc/bin to /usr/bin...

km4ack, 2022-01-22:
Finally got around to testing and noticed that if a GPS isn't connected, the script will hang. Can you add a timeout and echo out "NO GPS" if a GPS isn't attached to the Pi?

starbasessd, 2022-01-22:
Sure.

starbasessd, 2022-01-22:
On my systems, whether acting as a host for the GPS, or showing the Gridsquare from another host, it doesn't 'hang', it just shows (blank) if there isn't any data from a GPS. Is this why you want it to show 'NO GPS'?

km4ack, 2022-01-22:
I am working on something outside of BAP. This is the code I am using:
```
#!/usr/bin/env python3

from gpsdclient import GPSDClient
import maidenhead as mh

client = GPSDClient(host="localhost")

for result in client.dict_stream(convert_datetime=True):
    if result["class"] == "TPV":
        lat = result.get("lat", "n/a")
        lon = result.get("lon", "n/a")
        gridsquare = mh.to_maiden(lat, lon)
        print(gridsquare)
        break

```

When I run the script from the command line without a GPS connected to the pi, it will hang and doesn't return to prompt in the terminal. I assumed it would do the same thing if writing the grid to the text file.

starbasessd, 2022-01-22:
OK, I see the problem, occurs in latest version, too, if run separately. Will advise change.

starbasessd, 2022-01-22:
OK, Version 0.9.10 of instructions, including an updated PyGetGridsquare.py script.
I wrote the previous version forgetting the full functions of the If loop and For loop in Python.
I am also forcing "No GPS" if no local or network GPS Service found
[PyGetGridsquare.txt](https://github.com/km4ack/pi-build/files/7918689/PyGetGridsquare.txt)
.

km4ack, 2022-01-22:
That Works! Thanks!!

km4ack, 2022-01-22:
I spoke too soon. Now when I have a GPS connected and it has a lock, I am getting "NO GPS" On a positive note, it doesn't hang.

starbasessd, 2022-01-22:
Checking.

starbasessd, 2022-01-22:
Building Pi3B+, Bullseye 10/30/2021, fully updated. Using install instructions 1/22/2022 for the PyGetGridsquare.py, will report shortly.

starbasessd, 2022-01-22:
OK, running into a brick wall. 
I can check for /dev/ttyACM0 but not all USB GPSs connect there. 
I can check for gpsd, but my primary use case uses a network server, so it wouldn't find gpsd locally.
Almost everything I can check for seems to be a 1-up kind of thing. 
If you want to drop me an email, can we discuss your use case?

km4ack, 2022-01-22:
I like the python code and actually prefer it over ruby if it will function the same way with Conky. Currently the ruby script is reporting either the gridsquare or "no gps" to text file which is then displayed by conky. I would like to see the same behavior with the python script thus giving the operator some sort of indication if they either forgot to plug in the GPS or if the GPS doesn't have a lock.

starbasessd, 2022-01-22:
One issue is this allows both local and network sources, where Ruby didn't. [Thinking Hard](https://media.tenor.co/images/d638daabb115a4669c6811d7bc4b2967/tenor.gif).

km4ack, 2022-01-22:
Yep. I like that it will work over the network. I am not opposed to a two script solution like I currently use with the ruby script. If you can just get it to break out of the script if there's no GPS data available. That way it wouldn't have to record "no gps". If it simply returns empty, I can work with it.

km4ack, 2022-01-22:
This may be useless but there is a TPV object called "mode". Could you read it first to determine if it has a fix or not and exit out if no fix is found? Maybe exit if mode doesn't equal 3? reference - [https://gpsd.gitlab.io/gpsd/gpsd_json.html](https://gpsd.gitlab.io/gpsd/gpsd_json.html)

starbasessd, 2022-01-22:
The logic keeps falling back to testing for the GPSd service whether local or network. 
I think I found a way, minor re-write I think. more later.

starbasessd, 2022-01-22:
OK. Skull sweat. Version 0.9.11 with new (hopefully more robust version of PyGetGridsquare.py). 
Still writes gridsquare to /run/user/1000/gridinfo.txt and now "NO GPS" as appropriate.
Not as clean as before, but should be able to follow it. Had to break up client line.
[PyGetGridsquare.txt](https://github.com/km4ack/pi-build/files/7919336/PyGetGridsquare.txt)

starbasessd, 2022-01-22:
Don't ask for much, huh? LOL :)
I'll see what I can do for NO FIX...

starbasessd, 2022-01-23:
@km4ack 
You ask, you receive:
[PyGetGridsquare.txt](https://github.com/km4ack/pi-build/files/7920613/PyGetGridsquare.txt)
Reports "NO GPS" if it doesn't see a gpsd service running at host
Reports "No FIX" if TPV mode = 1 or 2
Reports gridsquare if it has a GPS service with a 3D fix.
Let me know... Looking for ibuprofen for head ache... grin

km4ack, 2022-01-23:
I wish I could count the number of "small" requests from others that had me beating my head against a wall for days. I'll give it a test run in a bit and let you know what I find. Thanks for the work on this!

starbasessd, 2022-01-23:
Helps me learn more about python, so not all bad.

km4ack, 2022-01-23:
I hate to be the bearer of bad news :-( 

GPS plugged in with fix - Script is good
GPS plugged in with no fix - Script errors or hangs as before
GPS not plugged in - Script hangs as before

I started with the GPS unplugged. Then I tried plugging in the GPS and running the script before it had a fix. Last thing I tried was waiting for the GPS to get a fix. 

Note: I am only waiting about 15-20 seconds before killing the script. Should I wait longer?

With the GPS plugged in but no fix I received these errors:
**Traceback (most recent call last):
  File "./pygridsquare", line 41, in <module>
    findgs()
  File "./pygridsquare", line 29, in findgs
    gridsquare = mh.to_maiden(lat, lon)
  File "/usr/local/lib/python3.7/dist-packages/maidenhead/to_maiden.py", line 23, in to_maiden
    a = divmod(lon + 180, 20)
TypeError: can only concatenate str (not "int") to str**

starbasessd, 2022-01-23:
OK, several issues:
You are not the bearer of bad news, yet, grin.

The problem with 'GPS not plugged in - Script hangs as before' may be that gpsd is installed wherever, is discovered on port 2947, but doesn't realize there isn't a GPS connected. This will entail checking the GPSd config file, and checking for the device. 
This won't work on a connection to a network server. 
This may be more effort than what it's worth. 
For Sanity check, IMO GPSd shouldn't make itself available if there isn't a GPS it can talk to.  
This program assumes you wouldn't be using it if: 
 > You don't have any GPS receivers (network or local) running (Should come back and say "NO GPS")
 > Your server (network or local) you want use is isn't available for whatever reason

Will check the Plugged in but no fix errors. I wonder why it thinks you are trying to concatenate when it looks like it is trying to add 180 to the reported longitude... 

I'll try a new setup here in a bit. Gotta dig out another USB GPS.

starbasessd, 2022-01-23:
What do you have for /etc/default/gpsd ?

km4ack, 2022-01-23:
#The -b switch was added per https://github.com/km4ack/pi-build/issues/144
#km4ack 24SEPT2020

# Default settings for the gpsd init script and the hotplug wrapper.

# Start the gpsd daemon automatically at boot time
START_DAEMON="true"

# Use USB hotplugging to add new USB devices automatically to the daemon
USBAUTO="false"

# Devices gpsd should collect to at boot time.
# They need to be read/writeable, either by user gpsd or the group dialout.
DEVICES="/dev/serial/by-id/usb-u-blox_AG_-_www.u-blox.com_u-blox_7_-_GPS_GNSS_Receiver-if00"

# Other options you want to pass to gpsd
GPSD_OPTIONS="-n -b"

starbasessd, 2022-01-23:
For DEVICES= I use /dev/ttyACM0
for GPSD_OPTIONS= I use "-n -G"
Your GPSD_OPTIONS negates read/writeable...
Hmmm...

km4ack, 2022-01-23:
the -b switch is due to this issue https://github.com/km4ack/pi-build/issues/144

km4ack, 2022-01-23:
I changed my GPSD_OPTIONS to match yours and get the same results. The /dev/serial/by-id is symlink to /dev/ttyACM0 so that shouldn't matter.

starbasessd, 2022-01-23:
I'm doing an almost complete re-write (at least as far as logic is concerned) will be back in a bit...

starbasessd, 2022-01-23:
Changing modes: Date night for son & dil, I get to watch the grandsons, so resuming tomorrow...

[14 more replies]