km4ack/pi-build #337: Chirp will not build on Bullseye

km4ack, 2021-11-11:
Need to revert to getting chirp from apt until I can figure out how to build from source or build using the flatpak. Currently python-serial and python-gtk2 aren't available on Bullseye. I suspect python-gtk3 may be needed but it isn't currently available either in bullseye.

starbasessd, 2021-11-11:
Interesting stuff.
http://www.peteronion.org.uk/PyGobjectGtk+3/PyGtk.html
I am also looking at the issue.

starbasessd, 2021-11-11:
GTK3+ is installed by default in RaspberryPiOS 11 Bullseye. See:
https://downloads.raspberrypi.org/raspios_armhf/release_notes.txt
There are 'work arounds' to get pip2 for python2 installed. I was unable to get flatpak to work due to it being X64, not any version of ARM... 
Writing up a suggestion.

starbasessd, 2021-11-12:
I can resolve every issue except one, so far, installing chirp in Bullseye on Pi.
sudo apt install libgtk2.0-dev
sudo apt-get install python2 python-dev-is-python2 -y
sudo curl https://bootstrap.pypa.io/pip/2.7/get-pip.py --output get-pip.py
sudo python2 get-pip.py
sudo pip2 install serial pyserial
sudo apt-get install python-gtk2-dev
at that point, running chirpw, I get an error from Python2 stating unable to load module gtk. 
So far, unable to find any replacements for the python2-gtk module for Bullseye.
Traceback (most recent call last):
  File "./chirpw", line 131, in <module>
    from chirp.ui import mainapp
  File "/home/pi/chirp-daily-20211105/chirp/ui/mainapp.py", line 26, in <module>
    import gtk
ImportError: No module named gtk

km4ack, 2021-11-16:
will revert to installing with apt until this is resolved.

km4ack, 2022-10-19:
issue resolved in 3.3.0