km4ack/pi-build #338: hamlib (rigctl) won't install on Bullseye SOLVED

km4ack, 2021-11-11:
libusb-1.0 not available in bullseye. Need to compile from source with:
```
sudo apt install -y libusb-dev libudev-dev
cd ~/Downloads
wget https://github.com/libusb/libusb/releases/download/v1.0.23/libusb-1.0.23.tar.bz2
tar -xvf libusb-1.0.23.tar.bz2
cd libusb-1.0.23/
./configure
make
sudo make install
sudo ldconfig
```
now run hamlib install as before with BAP

dafula, 2021-11-11:
Hi!
Just to clarify - there is a libusb-1.0 in Bullseye.
You can install the library and the header files via
apt install libusb-1.0-0 libusb-1.0-0-dev
Then the compiling of Hamlib 4.3.1 is without errors.
73! LZ1DAF

km4ack, 2021-11-11:
> Hi! Just to clarify - there is a libusb-1.0 in Bullseye. You can install the library and the header files via apt install libusb-1.0-0 libusb-1.0-0-dev Then the compiling of Hamlib 4.3.1 is without errors. 73! LZ1DAF

Thanks for the heads up. I overlooked the name change from 1.0 to 1.0-0

commit d98e2723 (2021-11-15): fix #338
closed 2021-12-05 by km4ack