km4ack/pi-build #349: Add CQRLOG

km4ack, 2021-11-18:
This app was removed in 3.2.0 because it couldn't be build from source and installing with apt also installed hamlib (rigctl) which broke the version built from source.

murtrah57, 2021-12-12:
So at this time under Bullseye there is no way to have Build a Pi working with a logging program since CQRLOG is removed? I guess I should go back to Buster and wait a while. Right now BAP and bullseye is only for non-critical experimentation.

km4ack, 2021-12-12:
> So at this time under Bullseye there is no way to have Build a Pi working with a logging program since CQRLOG is removed? I guess I should go back to Buster and wait a while. Right now BAP and bullseye is only for non-critical experimentation.

PyQSO and HAMRS are both still available for logging. Hopefully we can add CQRLOG back in the near future.

murtrah57, 2021-12-13:
HAMRS will not launch. I tried PyQSO but falls short on features compared to CQRLOG. I will go back to Buster until things settle down a bit with Bullseye. Thank you for what you do. I'll keep my eye out for info that the CQRLOG issues have been resolved.

km4ack, 2022-01-10:
Here's the solution that works on my Pi 4 running Bullseye if anyone would like to test. First update the repository with 
```
sudo apt update
```
Now install the dependencies. 
```
sudo apt-get install -y lcl-2.0 lazarus-2.0 fp-compiler-3.2.0 fpc-3.2.0 libssl-dev git build-essential 
libmariadb-dev-compat lazarus-ide lcl lcl-gtk2 lcl-nogui lcl-units lcl-utils lazarus lazarus-doc lazarus-src 
fp-units-misc fp-units-rtl fp-utils fpc fpc-source libssl-dev default-mysql-client-core default-mysql-server 
default-mysql-server-core gawk libcgi-fast-perl libcgi-pm-perl libconfig-inifiles-perl libdbd-mysql-perl 
libdbi-perl libencode-locale-perl libfcgi-perl libhtml-parser-perl libhtml-tagset-perl 
libhtml-template-perl libhttp-date-perl libhttp-message-perl libio-html-perl liblwp-mediatypes-perl 
libmariadb-dev-compat libnetpbm10 libreadline5 libterm-readkey-perl libtimedate-perl liburi-perl 
socat xplanet xplanet-images 
```
With the dependencies installed we can now build cqrlog
```
cd ~
git clone https://github.com/ok2cqr/cqrlog.git
cd cqrlog
make
sudo make install
```

gm1ymi, 2022-01-16:
If you copy/paste the dependencies above. First paste them into a text editor and make sure there are no extra line-breaks. Otherwise you'll be missing a bunch of dependencies. It'll look like CQRLOG has installed but mysql will be missing and it'll complain when it tries to create a new dB on startup.

km4ack, 2022-02-05:
added to v3.2.2

closed 2022-02-05 by km4ack