km4ack/pi-build #350: WSJTX/JS8Call Memory Issue

km4ack, 2021-11-28:
It was reported that WSJTX and possibly JS8Call will not build on a Pi 3 unless we increase the swap memory. Need to test and confirm.

graycott, 2021-11-30:
Bullseye and build a pi.
Jason, I can confirm that both WSJTX and JS8Call do not want to be installed on the Pi 3b plus. The swap memory and the cpu are running at max capacity. When this occurrs the system just slows down to a crawl.

km4ack, 2021-11-30:
> Bullseye and build a pi. Jason, I can confirm that both WSJTX and JS8Call do not want to be installed on the Pi 3b plus. The swap memory and the cpu are running at max capacity. When this occurrs the system just slows down to a crawl.

Thanks for the report!

graycott, 2021-12-01:
Jason,I increased the swap memory to 500M , and was able to complete the build.I did notice , a few times, the gauge on conky did hit the max. a couple of times. To me, this means 500M is about the ball park figure.I have successfully installed Js8 call and wsjtx, however pat winlink is not working. No transmitt ptt via cat, and no transmitt AUDIO to the FT991A. The GUI displaying the Rx signal  audio  ok, after setting the pat config  to  .....plughw:3,0 playhw:3,0.  Flrig seems to be working fine. Any ideas?Graham VK4GRASent from Samsung tablet.

"target": "https://github.com/km4ack/pi-build/issues/350#issuecomment-983027827",
"url": "https://github.com/km4ack/pi-build/issues/350#issuecomment-983027827",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]

closed 2021-12-05 by km4ack