km4ack/pi-build #351: add wsjtz

km4ack, 2021-11-30:
I have been using WSJT-Z for a while now. It has some very good features that can assist with off grid operations. For example filtering, can limit the stations that you see on a small screen or allow automatic communication with specific stations. It’s also great for field contest operation.

 

I would like to suggest that you add it to “build-a-pi” as an option.

 

Thanks for all that you do for the Ham community.

 

Richard Burgan

JS6TRQ / WC8J

Okinawa, Japan

NA8X, 2022-02-19:
This would be an awesome option for build-a-pi.  I recently upgraded to Raspbian 11 and am having difficulty getting wsjtz working again.  Had it working great on buster.  So far no love on bullseye.  I have read the documentation, but I keep getting an installation failed message that a needed package was not found.

SpudGunMan, 2022-10-11:
```
BUILDDIR=$HOME/.bap-source-files
#depends hamlib
#ran wsjtx installer first to get the required build enviroment setup (need copy here)
#this will not install as it will overwrite wsjtx this will compile and can be ran from there

WSJTPKG=wsjtz-2.5.4-1.22.zip

cd ${BUILDDIR} || return
mkdir wsjt-z
wget --tries 2 --connect-timeout=60  https://sourceforge.net/projects/wsjt-z/files/Source/$WSJTZPKG
unzip $WSJTZPKG
rm $WSJTZPKG
mv wsjtx src
cmake -DWSJT_SKIP_MANPAGES=ON-DWSJT_GENERATE_DOCS=OFF ./src
cmake --build .
#sudo cmake --build . --target install
#sudo cmake --build . --target uninstall

cat >wsjt-z.desktop <<EOF
[Desktop Entry]
Name=wsjt-z
GenericName=wsjtx-z
Comment=Popular Z Fork of wsjtx
Path=${BUILDDIR}/wsjt-z
Exec=wsjtx
Terminal=false
Type=Application
Categories=Network;HamRadio;
EOF

sudo mv wsjt-z.desktop /usr/share/applications/

```