km4ack/pi-build #352: Expanding the use of EES

KD7UM, 2021-12-04:
This is not an issue but an enhancement request.  It would be useful to be able to use the EES for emails between users at the remote site.  If we had a domain name that would keep the messages local and not require a review and approval of every message by the sysop the Pi could act as a email server for users connected to the Wifi.

Perhaps this is best done by a prebuilt package, if so great.  Things like shelter operation or a field day situation would allow emails to pass between users without being sent for www distribution over radio.

Thanks
Dave - KD7UM

km4ack, 2021-12-04:
Great suggestion! The reply is rather lengthy. I did it in video format so save a bit of time. [https://youtu.be/a5JSq685STA](https://youtu.be/a5JSq685STA) The install script for PHPBB that I mention in the video can be found [here](https://github.com/km4ack/pi-scripts/blob/master/install-phpBB)

closed 2021-12-04 by km4ack

KD7UM, 2021-12-04:
Nice video and super-fast reply.  What you detailed is what I thought might have to happen.  I am looking for a small router that will run on 12VDC for sharing files and movies off a Raspberry Pi NAS and Media server when away from the Internet.  It would not take any more hardware to load PHPBB on the same Pi.

 

Back in the hardware TNC days we had a BBS in every TNC that did a lot of what we needed in this request.  Unfortunately, the TNCs only worked with a RF interface.  A PHPBB would take very little compute power and would be a good application for a Pi3 or even a Pi zero.  You just need a little storage and a Wifi interface.

 

Thanks again.

starbasessd, 2021-12-04:
@KD7UM 
Pi3B+ and Pi4s have 5GHz Wifi (11ac) capability built in. Enough bandwidth for non-BlueRay streaming (I build Pis for car entertainment servers for grandkids and neighbor's kids) I have my 'main' Field Pi set up as a router already (DHCP/DNS/NTP/GPS/Email/NAS/etc) and it works well for Field Day exercises (camping/etc). I have several Belkin 5 & 8 port switches for Ethernet connections (12v powered), and the built in wifi as an AP/Wifi router. It can be done fairly easily, and possibly as part of BAP. Going to write something up for it if you or @km4ack are interested for either Buster or Bullseye, or stand alone without BAP. As always, I'd give unlimited permission to use as needed.

km4ack, 2021-12-04:
@KD7UM Here are my personal notes for setting up PHPBB for offline use. They are a couple of years old now but should be close enough to point you in the right direction:

phpBB settings

Access ACP

From left panel

Manage Groups

	Registered users - settings
		max number of allowed messages - 0
	New Users
		Allow private messages

Manage Forums
	Create New Forums as Needed
	Must copy permissions for it to be visible 
	

Attachment Settings - Update max allowed. All in MB
	2500
	10
	10
	10
	10
	Allow attachments in private messages
	Change allowed extension groups

Board Features
	Allow attachments in private messages - YES

Private Message Settings
	Max Private msgs per box - 250

User Registration Settings
	New Member Post Limit - 0
	Enable spambot - NO

Security Settings
	Check email MX record - NO

Notes below came from another help file of mine

A FEW CRITICAL SETTING IF USING OFFLINE

ACP=Admin Control Panel

ACP>Security Settings
Set check email domain for valid MX record to NO

ACP>User Registration settings
Set Enable Spambot Countermeasures to NO

ACP under permissions tab Choose User roles
set standard & newly registered to be able to send private message

ACP under posting tab choose manage attachment extension groups
be sure to allow documents Also note that under “Manage Attachments” you
can delete attachments if storage is running low.

ACP>Attachment Settings
Set total attachment quota (applies to whole board. I choose 10000 MiB which is 10GB)
Set Max file size allowed (I chose 25 MiB)
Set Max file size messaging (I chose 25 MiB)
Set number of attachments for post & private messages ( I Chose 5)

ACP>Private message setting
set max to 500

ACP under permissions tab choose forum permissions
put checkmark in all forums box/click submit
in groups window choose newly & registered/click edit permissions
change each to be standard+polls & click apply all permissions at bottom of page

KD7UM, 2021-12-04:
I would love to see how you did all of this.  I have a 2TB NAS Rpi server built using Open Media Vault.  I would like to layer a media server for times when we are away from home but it might be connected while at home.  When away from the home I need a WiFi router as you detailed below.

 

Jason has PHPBB running in his home.  I think it is also running his EES and might be tied in his VOIP radio network that he has also.

 

No rush on the writeup.  This is a busy time of the year but I am loving the Pi.  When I was a kid in college in 1969-1972 we got to work a little with a UNIX machine running on a PDP 10.  The textbook was all photocopied pages from AT&T materials.  The university was one of the original ARPA net schools (University of Utah).  That is how old I am!

 

Thanks for posting your response.  I appreciate it.  I have worked the past day with Jason killing a bug in his Emergency Email Server under Bullseye.  It is all working now.  He is great to work with.

 

Merry Christmas and Happy Holidays

Dave – KD7UM

starbasessd, 2021-12-05:
If you want to take this offline, feel free to contact me at kd9efv at gmail dot com. I have a write up for the car Pi video server, and am more than happy to share. I have 'most' of a doc for the email server, some of the functionality includes email, forums, task management, calendar (g*d forbid an event lasts longer than a few days), file transfers, user self registration, ability to send to outside email (can't receive, though, from the world, yet) lots of other stuff...

km4ack, 2021-12-05:
The PHPBB install script that I mentioned before does work after one minor mod. On [line 17](https://github.com/km4ack/pi-scripts/blob/master/install-phpBB#L17) you need to change php7.3 to php7.4 before running the script.

I just ran a quick test install and it worked with only the one mod made to the script.

On another note, I would be interested in seeing the notes from @starbasessd on setting up the email server. I am specifically interested in using it on a local area network and NOT connected to the internet.

starbasessd, 2021-12-05:
Which environment(s) do you prefer? Pi3B+ or Pi4-4GB? Buster or Bullseye? Lite or Desktop?

km4ack, 2021-12-05:
Pi4-4GB on Bullseye with the desktop. It might be something I can incorporate into Build a Pi in the future or at least create an install script similar to what I did with the PHPBB install script.

starbasessd, 2021-12-05:
No problem. I just threw up a Pi3B+, Buster, Desktop (custom built using pi-gen), to confirm instructions. Will have you some Standard Image Bullseye on the Pi4-4GB soon. I use 64GB USB3 for boot drives/operations, does that present a problem? (My available Pi4-4GB has the SDCard slot blocked by the case...)

km4ack, 2021-12-05:
Is this an image file or instructions to install? The USB won't be a problem if it is just instructions to build.

starbasessd, 2021-12-05:
Instructions to install. Does work on image for SDCard too. I've burned out way too many SDCards in the past 5 years...

km4ack, 2021-12-05:
Instructions would be perfect :-)

KD7UM, 2021-12-05:
I love where this is going.  Thanks to both of you for your work.  I am not a novice at Unix but there is so much I don’t know.

Dave