km4ack/pi-build #353: EES post to outbox on Bullseye and Pat 0.12 not working

KD7UM, 2021-12-04:
I initially thought this was a CRONJOB issue of not posting the EES emails to the PAT outbox but I believe it is a bug in the movetopi.html script referencing a bad location for the outbox.

The EES email is created is in /var/www/html/emails   

The mv command to post to the PAT outbox is referencing a folder that isn’t on Bullseye-PAT.  The old system had a folder named  home/pi/.wl2k.  The new PAT implementation has no such folder therefore the move to outbox doesn’t work.  I am not sure where the PAT In and Out boxes are in the Bullseye version of PAT.

The command mv /var/www/html/emails/*.b2f    $HOME/.wl2k/mailbox/$MYCALL/out is not a good location on PAT 0.12

km4ack, 2021-12-04:
This is an issue but not a Bullseye specific issue. This could happen on any Pi OS. Pat v0.12.0 changed the location of the Pat Mailbox files in addition to where the Pat Config file lives. I updated Pat Menu to accommodate the changes. Will get the EES updated ASAP.

km4ack, 2021-12-04:
Think I have the updates made to work with Pat 0.12. There isn't an automated process to update. You will have to do it manually. @KD7UM run the following:

```
cd ~/Downloads
wget https://raw.githubusercontent.com/km4ack/EES-LITE/dev/autopostreply
wget https://raw.githubusercontent.com/km4ack/EES-LITE/dev/movetopat
chmod +x autopostreply movetopat
sudo chown root:root autopostreply movetopat
sudo mv autopostreply movetopat /var/www/html/
```

Please let me know that it is working ok after you install & test.

KD7UM, 2021-12-04:
Close but not quite.  The movetopat html is putting the email into:  

 

It needs to go into /home/pi/.local/share/pat/mailbox/KD7UM/out    Setting the value of MYCALL didn’t work and I noticed it is different than the old BAP html

 

Did not try a reply since I couldn’t send the email that was moved to the wrong place.

KD7UM, 2021-12-04:
The problem might be in setting the value of CONFIG which messes up MYCALL.

km4ack, 2021-12-04:
Run these two commands please:

```
CONFIG=$XDG_CONFIG_HOME/pat/config.json
awk '/"mycall": "/{print}' $CONFIG |grep -o -P '(?<=mycall": ").*(?=",)' | tr '[:lower:]' '[:upper:]'
```

The first command will return nothing as it is setting the variable. The second command should return your call sign.

In your earlier post, you said it "The movetopat html is putting the email into:" but nothing was shown afterwards to indicate where it actually put the message.

KD7UM, 2021-12-04:
The set of commands did return an uppercase KD7UM callsign.

 

The email gets moved from EES to /home/pi/.local/pat/mailbox    It is setting with the folder for KD7UM, this is what the screen shot shows.  It needs to go to /home/pi/.local/pat/mailbox/KD7UM/out    Since the KD7UM did not get filled in the mv command it dropped in one level higher than it should.  You can see the KD7UM folder and a file named :out” in the mailbox folder.

 

Sorry for the confusion.

closed 2021-12-09 by km4ack