km4ack/pi-build #359: BAP update deleted xlog

wb8erj, 2021-12-10:
Hello,

I recently updated to Bullseye / BAP latest version. During the BAP install I noticed there was no selection for Xlog. Which is fine if you want to remove it. I just did an apt-get install xlog and all was well.

Until a few says later when I ran the BAP update and it removed my manual install of Xlog. I know not many people use it, and that is probably why it was removed from the BAP install choices, no big deal. 

For me I like Xlog because it supports the winkeydaemon as well as two optional logging fields for POTA for their SIG_INFO and MY_SIG_INFO fields. BTW, this is just an explanation as to why I use xlog, not asking for it to be added to the list, I can do that no problem. However when I run a BAP update if BAP didn't remove xlog that would be great.

-- Mike

km4ack, 2021-12-10:
Hey Mike,

xlog was one item that I chose to remove from BAP BAP has no capability of removing any application yet. Did you try running xlog from the command line or running the whereis command? 
`whereis xlog`

km4ack, 2021-12-10:
It just dawned on me as to what may be happening. Some applications install hamlib with apt and this conflicts with hamlib being built from source with BAP. To combat this conflict, there is a line of code in BAP that removes hamlib that was installed with apt. See [this line](https://github.com/km4ack/pi-build/blob/master/update#L618) This may also be uninstalling xlog since xlog may depend on libhamlib4 being present.  You could comment out that line of code by place a "#" in front of it. Doing so may cause conflicts though with the system.

wb8erj, 2021-12-10:
Thanks for responding. Well that makes sense. with the exception that I was
watching the screen scroll by and saw that it had removed xlog. It was
after BAP was done compiling.

I did try to run Xlog by clicking on the desktop icon that I had made (it
always worked before..) and it said something to the effect that the
program was not there. Unfortunately I just re-installed xlog. You are
right, I should have done what you suggested.

-- Mike WB8ERJ

Geeze! What is he up to now?
https://mikestechblog.com

wb8erj, 2021-12-10:
OK thank you! Yes I use hamlib with xlog. Well I don't want to upset my BAP
apple cart, so its really NBD to just reinstall.

Thanks tor your time looking into this, and your excellent work!

-- Mike WB8ERJ

Geeze! What is he up to now?
https://mikestechblog.com

closed 2022-08-31 by km4ack