km4ack/pi-build #361: Install command does not work

murtrah57, 2021-12-12:
When I run the command line you give for installation it results in errors about directories not existing. Looking at the files system shows folder created with no name. What am I doing wrong?

murtrah57, 2021-12-12:
pi@raspberrypi:~ $ git clone https://github.com/km4ack/pi-build.git \ $HOME/pi-build && bash $HOME/pi-build/build-a-pi
Cloning into ' /home/pi/pi-build'...
remote: Enumerating objects: 2127, done.
remote: Counting objects: 100% (1067/1067), done.
remote: Compressing objects: 100% (727/727), done.
remote: Total 2127 (delta 685), reused 698 (delta 338), pack-reused 1060
Receiving objects: 100% (2127/2127), 754.89 KiB | 8.03 MiB/s, done.
Resolving deltas: 100% (1347/1347), done.
bash: /home/pi/pi-build/build-a-pi: No such file or directory
pi@raspberrypi:~ $

murtrah57, 2021-12-12:
[image]

murtrah57, 2021-12-12:
I resolved the issue by watching the install video. Not sure why but the text command on github has a \ in the middle. In your video that \ is not there. When I run the command without the \ all is well. Not sure how the \ got into your text on the site but you may want to fix that to save others much wasted time and fixing of their file system.

km4ack, 2021-12-12:
You paste the command into the terminal as two lines but at the same time.
git clone https://github.com/km4ack/pi-build.git \
  $HOME/pi-build && bash $HOME/pi-build/build-a-pi

The "\" at the end of the first line tells the system to expect another line of code. This simply makes it easier to read than one long line of code. You got the error because you put both lines of code on the same line instead of two separate lines.

closed 2021-12-12 by km4ack