km4ack/pi-build #363: BAP 3.2.0 says YAAC needs upgrading continuously

KD7UM, 2021-12-24:
I did a BAP update on a Pi today running Bullseye.  It used BAP 3.2.0 and I selected YAAC to be updated.  It ran, I rebooted, and then ran BAP again and it said YAAC needed updating.  The version I am running is YAAC 1.0-beta167  13-Dec-2021 which sounds like a pretty current version.

I am not sure how BAP is thinking it isn't current or if BAP is correct, the current version is not installing

Merry Christmas - no rush
Dave - KD7UM

km4ack, 2021-12-25:
Thanks for the report Dave. Looks like the output from YAAC changed a bit. I'll get it fix in the next release.

Change line [353](https://github.com/km4ack/pi-build/blob/master/app-check#L353) of the app-check script to be:
`CURYAAC=$(java -jar ${HOME}/YAAC/YAAC.jar -version | grep beta | sed 's/.*beta//;s/(.*//')`

km4ack, 2022-01-06:
This one has me stumped at the moment. If I run the app-check script alone, it returns the correct results for YAAC in the /run/user/1000/bapupdate.txt file. However, when using the BAP update tool which calls the app-check script it will not return the current version of YAAC installed. As a result, it always says "needs update"

KD7UM, 2022-01-06:
I am a programmer but not for anything in this century, but I wonder if you could display some of the “environment” variables that get set before and after calling the various scripts.  It seems like something is not getting set right when calling the apt-check script.

 

If I can help test, I would like to

73, Dave

KD7UM, 2022-01-06:
Would it help if you displayed CURYACC and LATESTYACC and halt the script just before the if statement?  Also display the results from echo (($(echo "${LATESTYAAC} ${CURYAAC}" to see if something has slid over a character or truncated something from comparing the two versions.

 

 

                CURYAAC=$(java -jar ${HOME}/YAAC/YAAC.jar -version | grep core | awk '{print $7}' | sed 's/(.*//;s/1.0-beta//')

                wget -q https://sourceforge.net/projects/yetanotheraprsc/files/YAACBuildLabel.txt -O /run/user/${UID}/latestyaac.txt

                LATESTYAAC=$(cat /run/user/${UID}/latestyaac.txt | sed 's/1.0-beta//;s/(.*//')

                if (($(echo "${LATESTYAAC} ${CURYAAC}" | awk '{print ($1 > $2)}'))); then

                        echo "YAAC=NEEDS-UPDATE" >> $UPDATEFILE

                else

                        echo "YAAC=is_latest_version" >> $UPDATEFILE

                fi

KD7UM, 2022-01-06:
We had a problem with an early BAP issue where you assumed ownership of root on a file when it wasn’t.  Could this be the same problem?

km4ack, 2022-01-07:
I have tried echoing out the variables. They are correct when I run the app-check script alone but the current version variable is empty when I call the app-check script from the update script. I don't think it is an ownership issue and I have also changed the script in my test to include the full path to each item. All of the other checks in the app-check script are working as expected when called from the update script.

km4ack, 2022-01-07:
I haven't gotten completely to the bottom but know where the conflict is occurring. If there are patches available, YAAC will not indicate correctly. If there are no patches available, YAAC will indicate correctly once line 353 of the app-check script is changed to `CURYAAC=$(java -jar ${HOME}/YAAC/YAAC.jar -version | grep beta | sed 's/.*beta//;s/(.*//')` What I still can't figure out is why the patch check in the update script is causing a conflict with checking the YAAC version. I have made the change to line 353 but still need to investigate why the conflict occurs when patches are available.

commit 2993bcad (2022-01-07): fix #363
closed 2022-01-08 by km4ack