km4ack/pi-build #365: WSJTX not building on Bullseye

gm1ymi, 2021-12-26:
rPI4 running bullseye.
seems to download wsjtx-2.5.3 but fails at that point.

--2021-12-26 07:16:39--  https://physics.princeton.edu/pulsar/k1jt/wsjtx-2.5.3.tgz
Resolving physics.princeton.edu (physics.princeton.edu)... 128.112.100.6
Connecting to physics.princeton.edu (physics.princeton.edu)|128.112.100.6|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 25599071 (24M) [application/x-gzip]
Saving to: ‘wsjtx-2.5.3.tgz’

wsjtx-2.5.3.tgz     100%[===================>]  24.41M  6.66MB/s    in 4.5s    

2021-12-26 07:16:44 (5.46 MB/s) - ‘wsjtx-2.5.3.tgz’ saved [25599071/25599071]

http://#000000: Invalid host name.
--2021-12-26 07:16:44--  http://wsjtx-2.5.2.tgz/
Resolving wsjtx-2.5.2.tgz (wsjtx-2.5.2.tgz)... failed: Name or service not known.
wget: unable to resolve host address ‘wsjtx-2.5.2.tgz’
FINISHED --2021-12-26 07:16:44--
Total wall clock time: 5.2s
Downloaded: 1 files, 24M in 4.5s (5.46 MB/s)
tar: #000000: Not found in archive
tar: wsjtx-2.5.2.tgz: Not found in archive
tar: Exiting with failure status due to previous errors
rm: cannot remove '#000000': No such file or directory
rm: cannot remove 'wsjtx-2.5.2.tgz': No such file or directory
/home/pi/pi-build/functions/additional.function: line 222: cd: too many arguments
CMake Error: The source directory "/home/pi/.bap-source-files/wsjtx-2.5.2.tgz" does not exist.
Specify --help for usage, or press the help button on the CMake GUI.
CMake Warning:
  No source or binary directory provided.  Both will be assumed to be the
  same as the current working directory, but note that this warning will
  become a fatal error in future CMake releases.

CMake Error: The source directory "/home/pi/.bap-source-files" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
Error: could not load cache
mv: cannot stat '/home/pi/.bap-source-files/wsjtx-2.5.3': No such file or directory
mv: cannot stat '#000000': No such file or directory
mv: cannot stat 'wsjtx-2.5.2.tgz/bin/*': No such file or directory
cp: cannot stat '/home/pi/.bap-source-files/wsjtx-2.5.3': No such file or directory
cp: cannot stat '#000000': No such file or directory
cp: cannot stat 'wsjtx-2.5.2.tgz/share/applications/wsjtx.desktop': No such file or directory
cp: cannot stat '/home/pi/.bap-source-files/wsjtx-2.5.3': No such file or directory
cp: cannot stat '#000000': No such file or directory
cp: cannot stat 'wsjtx-2.5.2.tgz/share/pixmaps/wsjtx_icon.png': No such file or directory
updating fldigi
updating flarq
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package 'libhamlib4' is not installed, so not removed
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

kultapanda, 2021-12-26:
looks like the curl command for pkg variable is pulling back the following:
wsjtx-2.5.3.tgz
#000000
wsjtx-2.5.2.tgz

Would it be acceptable to just add “| head” to that first curl command?

km4ack, 2021-12-26:
Yes. This [line](https://github.com/km4ack/pi-build/blob/master/functions/additional.function#L216) needs to be modified to be:
`curl -s https://physics.princeton.edu/pulsar/k1jt/wsjtx.html | grep .tgz | sed 's/.*="//;s/">.*$//' | head -1`
and it should work again. Looks like they have two different .tgz files on the page now which wasn't there when I wrote the initial code.

gm1ymi, 2021-12-26:
> Yes. This [line](https://github.com/km4ack/pi-build/blob/master/functions/additional.function#L216) needs to be modified to be: `curl -s https://physics.princeton.edu/pulsar/k1jt/wsjtx.html | grep [...]

Yes, I just added '| head -1 ' to additional.function and can confirm it now builds without issue. Thanks for the quick response. @kultapanda @km4ack

gm1ymi, 2021-12-28:
Seg faults on FT8 decode ... but don't think it's BAP related. There's numerous comments over on the wsjtx groups.io related to seg faults & rPI.
gdp result below:
Thread 1 "wsjtx" received signal SIGSEGV, Segmentation fault.
0xb58f8c40 in QtPrivate::qustrchr(QStringView, unsigned short) ()
   from /lib/arm-linux-gnueabihf/libQt5Core.so.5

KD7UM, 2021-12-31:
For those trying to fix this error themselves you need to edit /home/pi/pi-build/functions/additional.function
and change the line that reads:
FT8PKG=$(curl -s https://physics.princeton.edu/pulsar/k1jt/wsjtx.html | grep .tgz | sed 's/.*="//;s/">.*$//')
to 
FT8PKG=$(curl -s https://physics.princeton.edu/pulsar/k1jt/wsjtx.html | grep .tgz | sed 's/.*="//;s/">.*$//' | head -1)

commit f7b884ee (2022-01-06): fix #365
closed 2022-01-08 by km4ack