km4ack/pi-build #369: Mobilinkd TNC3 & Pat

KJ7RNY, 2022-01-10:
I am receiving an error message with Mobilinkd and Pat Winlink.  The error message "no AX.25 no ports configured" is displayed when trying to connect to RMS gateways.

When using Build-A-Pi on Rasbian Buster the TNC worked with Pat.  Since upgrading to Bullseye it no longer works.

I followed the instructions in the Pi Zero Project::Mobilinkd Winlink video.  Everything works from 8:20 to 13:10 in the video (bluetooth pairing, rfcomm bind command, kissattach command).  When I attempt the command 'sudo kissparms -c 1 -p wl2k' I receive the "no AX.25 no ports configured" error.

Current setup: Raspberry Pi 4, Rasbian Bullseye, KM4ACK build-a-pi, Mobilinkd TNC3, Icom IC-705.

The /etc/ax25/axports file reads: wl2k <callsign> 1200 255 7 Winlink

km4ack, 2022-01-12:
Have you tried starting the mobilinkd modem with Pat Menu? That feature is still in beta but most have reported success when using it.

KJ7RNY, 2022-01-13:
Yes.  I am on PatMenu v2.5.0.  I am beginning to suspect an issue with the TNC3 on Rasbian Bullseye.  I am able to make a RMS gateway connection using the TNC3 and Patmenu on a fresh install of Rasbian Buster and Build-a-pi.  When I use my current Rasbian Bullseye install with Patmenu the mobilink script completes but I am unable to make RMS gateway connections.

The mobilink script is successfully creating the /dev/rfcomm0 serial connection.  When the mobilink script completes I am opening the Pat GUI.  When I go to connect to a gateway this is what is in the log for Pat:

Log using Bullseye
2022/01/13 09:16:01 QSY failed: unable to get rig frequency: dial tcp 127.0.0.1:4532: connect: connection refused
2022/01/13 09:16:23 Connecting to KG7ARL-10 (ax25)...
2022/01/13 09:16:23 Unable to establish connection to remote: No AX.25 ports configured

Log using Buster
2022/01/13 11:57:49 Unable to get frequency from rig my-rig: dial tcp 127.0.0.1:4532: connect: connection refused.
2022/01/13 11:57:49 Starting HTTP service (http://127.0.0.1:8080)...
2022/01/13 12:03:53 Connecting to KG7ARL-10 (ax25)...
2022/01/13 12:04:38 Unable to establish connection to remote: Dial timeout
2022/01/13 12:13:13 Connecting to KG7ARL-10 (ax25)...
2022/01/13 12:13:58 Unable to establish connection to remote: Dial timeout
2022/01/13 12:14:50 Connecting to KG7ARL-10 (ax25)...
2022/01/13 12:14:53 Connected to KG7ARL-10 via WTANKS (AX.25)
[WL2K-5.0-B2FWIHJM$]
;PQ: 01520896
CMS via KG7ARL >
>FF
FQ
2022/01/13 12:15:02 Disconnected.

KJ7RNY, 2022-01-15:
Issue resolved with a fresh install of Raspian Bullseye and Build-a-pi.

closed 2022-01-15 by KJ7RNY