km4ack/pi-build #371: BAP v3.2.1 says PATMENU needs an update continually

KD7UM, 2022-01-11:
On two different Pi systems running BAP 3.2.1 gives me a message that PATMENU needs an update.  It is version 2.5.0 now.  I select it and it runs but never installs PATMENU.  A reboot and subsequent BAP Update gives me the same message to update PATMENU.

km4ack, 2022-01-11:
Run these two commands and see if it clears up
```
cd ~/patmenu2
git checkout master
```

KD7UM, 2022-01-11:
This is what I got:
pi@PI3BS:~ $ cd ~/patmenu2
pi@PI3BS:~/patmenu2 $ git checkout master
M	config
D	setup
Switched to branch 'master'
Your branch is up to date with 'origin/master'.
pi@PI3BS:~/patmenu2 $ 
----------------------
Now it says PAT-Menu is the latest version.  Should this be done to all systems that were updated in some earlier form?

km4ack, 2022-01-11:
It's left over from when pat menu was supporting both the old version and new version of Pat Winlink. I didn't notice it until after the last update went out and wasn't sure it would impact everyone. Often there are quirks in my Pi due to all of the various testing. I'll have to write some code into the update script to fix it in the next release. You should also see that Pat Menu is v2.6.0 when you open it now.

KD7UM, 2022-01-11:
Yes, Pat Menu is now v2.6.0  I did the same thing on another Pi also.  Ok on the update in the next version.

closed 2022-02-05 by km4ack