km4ack/pi-build #383: VNC Viewer not installing

km4ack, 2022-03-03:
realvnc.com has blocked curl requests on their site. Due to this block, it isn't possible for BAP to use the curl command to determine the latest version of VNC viewer. We can still download the viewer from their site using a wget command in BAP but this will require hard coding the version number into BAP. For now this is the approach that should be taken but support may be dropped in the future since this process can't be automated and may routinely break the BAP code.

starbasessd, 2022-03-03:
In order to give yourself some 'breathing room', yet give users a vnc viewer client, please consider gvncviewer, a simple VNC Viewer that works well for me, easily installed with 'sudo apt install gvncviewer', is fairly lightweight, and works with all the VNC Servers I've tested against. You could create a link in the start menu easily, too.

km4ack, 2022-03-17:
resolved in BAP 3.2.3

closed 2022-03-17 by km4ack