km4ack/pi-build #384: Could not resolve host: github.com

g4jkz, 2022-03-10:
New to pi-build and github platform.
I have followed all instructions in the video to the point of downloading the software but every time I try, I get the response as below.  I am using a Pi4 and the specified pi OS ...... any ideas would be most helpful ... convinced it's me at fault!

pi@raspberrypi:~ $  git clone https://github.com/km4ack/pi-build.git \
  $HOME/pi-build && bash $HOME/pi-build/build-a-pi
Cloning into '/home/pi/pi-build'...
**fatal: unable to access 'https://github.com/km4ack/pi-build.git/': Could not resolve host: github.com**
pi@raspberrypi:~ $

g4jkz, 2022-03-10:
> I assume your pi is connected to the internet? Run both of the lines below as one command:
> 
> ```
> git clone https://github.com/km4ack/pi-build.git \
>   $HOME/pi-build && bash $HOME/pi-build/build-a-pi
> ```
> 
> Copy/paste is your friend and be sure you include the "\" at the end of the first line.

Yes copied exactly as advised but still have the same result.  All linked via cable to the router and I am using the Pi directly not via SSH option - will this matter?

km4ack, 2022-03-10:
Being direct on the pi is fine. Try these commands **one at a time** and see if it makes a difference. If the git clone command fails, don't bother with the rest.
```
cd ~
git clone https://github.com/km4ack/pi-build.git
cd ~/pi-build
bash build-a-pi
```

g4jkz, 2022-03-10:
No .. alas it failed as previously on the first line.

km4ack, 2022-03-10:
There is either an issue with your internet connection or (less likely) and issue with your Pi OS image. I have verified that the commands work here. Are you able to open a web browser and navigate to https://github.com/km4ack ?

g4jkz, 2022-03-10:
Yes internet working okay as sourced info from your Github page okay.

g4jkz, 2022-03-10:
Thanks Jason ... I will look at the router rules in case anything is set to prevent running the commands.

g4jkz, 2022-03-10:
Unpacked okay, thanks but now saying ... no such file or directory ...

pi@raspberrypi:~ $ unzip ~/Downloads/pi-build-master.zip -d ~/pi-build
Archive:  /home/pi/Downloads/pi-build-master.zip
95746aad0a18a0f8ca0f2dd43fc5fbaf8c2cfdb4
   creating: /home/pi/pi-build/pi-build-master/
  inflating: /home/pi/pi-build/pi-build-master/.complete  
  inflating: /home/pi/pi-build/pi-build-master/.funfacts  
 extracting: /home/pi/pi-build/pi-build-master/.gitignore  
  inflating: /home/pi/pi-build/pi-build-master/README.md  
  inflating: /home/pi/pi-build/pi-build-master/app-check  
  inflating: /home/pi/pi-build/pi-build-master/bap-wallpaper.jpg  
  inflating: /home/pi/pi-build/pi-build-master/build-a-pi  
  inflating: /home/pi/pi-build/pi-build-master/changelog  
   creating: /home/pi/pi-build/pi-build-master/conky/
  inflating: /home/pi/pi-build/pi-build-master/conky/.conkyrc  
  inflating: /home/pi/pi-build/pi-build-master/conky/conky-7inch  
  inflating: /home/pi/pi-build/pi-build-master/conky/conky-large  
  inflating: /home/pi/pi-build/pi-build-master/conky/conky-logo.png  
  inflating: /home/pi/pi-build/pi-build-master/conky/conky-medium  
  inflating: /home/pi/pi-build/pi-build-master/conky/conky-small  
  inflating: /home/pi/pi-build/pi-build-master/conky/get-freq  
  inflating: /home/pi/pi-build/pi-build-master/conky/get-grid  
  inflating: /home/pi/pi-build/pi-build-master/conky/get-vfo  
  inflating: /home/pi/pi-build/pi-build-master/conky/grid  
  inflating: /home/pi/pi-build/pi-build-master/conky/setconky  
  inflating: /home/pi/pi-build/pi-build-master/diagnostic  
   creating: /home/pi/pi-build/pi-build-master/functions/
  inflating: /home/pi/pi-build/pi-build-master/functions/additional.function  
  inflating: /home/pi/pi-build/pi-build-master/functions/autohotspot  
  inflating: /home/pi/pi-build/pi-build-master/functions/base.function  
  inflating: /home/pi/pi-build/pi-build-master/functions/flsuite.function  
  inflating: /home/pi/pi-build/pi-build-master/functions/utility.function  
  inflating: /home/pi/pi-build/pi-build-master/logo.png  
  inflating: /home/pi/pi-build/pi-build-master/menu-update  
  inflating: /home/pi/pi-build/pi-build-master/patch-menu  
   creating: /home/pi/pi-build/pi-build-master/patch/
  inflating: /home/pi/pi-build/pi-build-master/patch/README.md  
  inflating: /home/pi/pi-build/pi-build-master/patch/test-patch  
  inflating: /home/pi/pi-build/pi-build-master/update  
pi@raspberrypi:~ $ bash ~/pi-build/build-a-pi
bash: /home/pi/pi-build/build-a-pi: **No such file or directory**
pi@raspberrypi:~ $

g4jkz, 2022-03-10:
'pi-build' directory has been created  ... but not the 'build-a-pi'

pi@raspberrypi:~ $ bash ~/pi-build/build-a-pi
bash: /home/pi/pi-build/build-a-pi: No such file or directory
pi@raspberrypi:~ $ cd pi-build-master
bash: cd: pi-build-master: No such file or directory
pi@raspberrypi:~ $ cd pi-build
pi@raspberrypi:~/pi-build $ cd build-a-pi
bash: cd: build-a-pi: No such file or directory
pi@raspberrypi:~/pi-build $

km4ack, 2022-03-10:
`bash ~/pi-build/build-a-pi`

km4ack, 2022-03-10:
If the previous command doesn't work, run:
```
cd ~/pi-build
ls
```
and post the results

g4jkz, 2022-03-10:
pi@raspberrypi:~ $ bash ~/pi-build/build-a-pi
bash: /home/pi/pi-build/build-a-pi: No such file or directory
pi@raspberrypi:~ $ cd ~/pi-build
pi@raspberrypi:~/pi-build $ ls
pi-build-master
pi@raspberrypi:~/pi-build $

km4ack, 2022-03-10:
ok. now run:
```
cd ~/Downloads
ls
```
and post the results.

g4jkz, 2022-03-10:
pi@raspberrypi:~/pi-build $ cd ~/Downloads
pi@raspberrypi:~/Downloads $ ls
boat-race-en-resources.zip  pi-build-master.zip
pi@raspberrypi:~/Downloads $

km4ack, 2022-03-10:
sorry. i missed one step in the previous set of commands. try:
```
cp -r ~/pi-build/pi-build-master/* ~/pi-build/
bash ~/pi-build/build-a-pi
```

g4jkz, 2022-03-10:
Okay, so following presented ..

#######################################
#  Updating repository & installing   #
#  a few needed items before we begin #
#######################################
Err:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease              
  Temporary failure resolving 'raspbian.raspberrypi.org'
Err:2 http://archive.raspberrypi.org/debian bullseye InRelease                 
  Temporary failure resolving 'archive.raspberrypi.org'
Reading package lists... Done             
W: Failed to fetch http://raspbian.raspberrypi.org/raspbian/dists/bullseye/InRelease  Temporary failure resolving 'raspbian.raspberrypi.org'
W: Failed to fetch http://archive.raspberrypi.org/debian/dists/bullseye/InRelease  Temporary failure resolving 'archive.raspberrypi.org'
W: Some index files failed to download. They have been ignored, or old ones used instead.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following package was automatically installed and is no longer required:
  libfuse2
Use 'sudo apt autoremove' to remove it.
The following NEW packages will be installed:
  yad
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 162 kB of archives.
After this operation, 512 kB of additional disk space will be used.
Err:1 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf yad armhf 0.40.0-1
  Temporary failure resolving 'raspbian.raspberrypi.org'
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/y/yad/yad_0.40.0-1_armhf.deb  Temporary failure resolving 'raspbian.raspberrypi.org'
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following package was automatically installed and is no longer required:
  libfuse2
Use 'sudo apt autoremove' to remove it.
The following additional packages will be installed:
  libjq1 libonig5
The following NEW packages will be installed:
  jq libjq1 libonig5
0 upgraded, 3 newly installed, 0 to remove and 0 not upgraded.
Need to get 355 kB of archives.
After this operation, 1,056 kB of additional disk space will be used.
Err:1 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf libonig5 armhf 6.9.6-1.1
  Temporary failure resolving 'raspbian.raspberrypi.org'
Err:2 http://raspbian.mirror.uk.sargasso.net/raspbian bullseye/main armhf libjq1 armhf 1.6-2.1
  Temporary failure resolving 'raspbian.mirror.uk.sargasso.net'
Get:3 http://raspbian.mirror.uk.sargasso.net/raspbian bullseye/main armhf jq armhf 1.6-2.1 [64.6 kB]
Fetched 64.6 kB in 41s (1,579 B/s)                
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/libo/libonig/libonig5_6.9.6-1.1_armhf.deb  Temporary failure resolving 'raspbian.raspberrypi.org'
E: Failed to fetch http://raspbian.mirror.uk.sargasso.net/raspbian/pool/main/j/jq/libjq1_1.6-2.1_armhf.deb  Temporary failure resolving 'raspbian.mirror.uk.sargasso.net'
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
/home/pi/pi-build/build-a-pi: line 194: yad: command not found
/home/pi/pi-build/build-a-pi: line 215: yad: command not found
/home/pi/pi-build/build-a-pi: line 194: yad: command not found
/home/pi/pi-build/build-a-pi: line 215: yad: command not found
/home/pi/pi-build/build-a-pi: line 194: yad: command not found
/home/pi/pi-build/build-a-pi: line 215: yad: command not found
/home/pi/pi-build/build-a-pi: line 194: yad: command not found
/home/pi/pi-build/build-a-pi: line 208: yad: command not found
pi@raspberrypi:~ $

g4jkz, 2022-03-10:
Hi Jason,  will have to return to this tomorrow as getting late now and don't want to tax you anymore tonight! .... thanks for your patience and help so-far, I will use another SD card with a fresh download and see if that resolves the github problem.  If not will retrace the above steps again and report back to you.
In the meantime, cheers for now .. Keith

km4ack, 2022-03-10:
You certainly have some issue with the internet connection. It is not only github that the pi can't reach as indicated by this error `E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/libo/libonig/libonig5_6.9.6-1.1_armhf.deb Temporary failure resolving 'raspbian.raspberrypi.org'`

starbasessd, 2022-03-10:
@km4ack may I suggest installing dnsutils (dig, nslookup) for diagnosis?
nslookup is a simpler report, and shows not only the dns address of the query, but what server is being used.
>sudo apt install dnsutils

For example:
- - - - -
nslookup google.com
Server:         192.168.x.y
Address:        192.168.x.y#53

Non-authoritative answer:
Name:   google.com
Address: 142.250.81.206
Name:   google.com
Address: 2607:f8b0:4004:814::200e
- - - - -
It helps in these kinds of situations, where something is set up wrong (static IP, gateway, dns server, etc...)

starbasessd, 2022-03-10:
nslookup raspbian.raspberrypi.org
Server:         192.168.2.103
Address:        192.168.2.103#53

Non-authoritative answer:
raspbian.raspberrypi.org        canonical name = mirrordirector.raspbian.org.
Name:   mirrordirector.raspbian.org
Address: 93.93.128.193
Name:   mirrordirector.raspbian.org
Address: 2a00:1098:0:80:1000:75:0:3

km4ack, 2022-03-10:
> @km4ack may I suggest installing dnsutils (dig, nslookup) for diagnosis? nslookup is a simpler report, and shows not only the dns address of the query, but what server is being used.
>
>
> For example: [...]

@starbasessd Not a bad suggestion. What is really odd is that he seems to be able to use the browser to navigate but nothing seems to work when called from the command line.

starbasessd, 2022-03-10:
Which browser? On what machine? Are you assuming he's using the Pi's browser? I never do (assume, that is)...

starbasessd, 2022-03-10:
Oh, and I forgot the most important thing: 

## REBOOT and test again....

km4ack, 2022-03-10:
Yes. I assumed (I know I shouldn't) that he was using the browser on the Pi. Since this was a default install, I would guess Chromium.

g4jkz, 2022-03-11:
Hi Jason,  after an afternoon of searching the web and trying out solutions I finally have the process loading now using the following recommendation which I am assuming replaces the name text for github (?).
I did initially try many changes on the Router config with no luck.
However it's been running now for 50 mins (having setup my options) and still looking good.  Thanks for all your help it's very much appreciated.

1.) Open file
  sudo nano /etc/resolv.conf

2.) Add the lines to the file:
  nameserver 8.8.8.8
  nameserver 8.8.4.4
    
1.) Open file
  sudo nano /etc/dnsmasq.conf
  
2.) Add the lines to the file:
  server=8.8.8.8
  server=8.8.4.4

Cheers ... 73 ... Keith (gm4jkz)