km4ack/pi-build #387: Is it possible to add GPSD support in Direwolf

wb8erj, 2022-03-15:
I would like to run direwolf with direct support of the GPS device. In doing a cursory search, adding gpsd and libgps-dev to the direwolf compile should do the trick.

Thanks for looking into this

-- Mike

km4ack, 2022-03-15:
Provided you install the GPS software with BAP, direwolf will be compiled with GPSD support. I just did a new build this afternoon with BAP 3.2.3 and direwolf has support for GPSD, hamlib, and cm108-ptt. You will need to choose hamlib during the build for this support to be included. When you start direwolf, the second line should read like this:
**Includes optional support for:  gpsd hamlib cm108-ptt**
assuming you chose GPS and hamlib during the build. 

If you didn't choose the needed extras during the first run, go back and install hamlib and the GPS software. Then on the next run, tell BAP to reinstall direwolf and you should be good.

wb8erj, 2022-03-15:
Ok thanks. I must hv missed a stepSent from my Verizon, Samsung Galaxy smartphone

"target": "https://github.com/km4ack/pi-build/issues/387#issuecomment-1067466733",
"url": "https://github.com/km4ack/pi-build/issues/387#issuecomment-1067466733",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]

closed 2022-08-26 by km4ack

yjf007, 2024-01-14:
hello，my friend，I having the same problem，Config file, line 6: The GPSD interface has not been enabled.
2024-01-14 14:20:56 Install gpsd and libgps-dev packages then rebuild direwolf.