km4ack/pi-build #395: BullsEye 64 Bit Issues

Mooman2022, 2022-04-03:
BullsEye 64 Bit Issues

FLRIG

    • When first opened FLRIG defaults to a frequency in the 20 meter band. A frequency I do not use. Does not match what the radio is set to.

    • Trying to re-configure from menu does not work – radio will not respond.

    • Have to physically remove the CAT cable and plug into a different USB port. Do another configure update from the menu, now the radio responds??

Main Issue:

    • Once I have FLRIG menu open the yellow “Split” lamp is on. Pushing it off turns it off and within 2 seconds lamp comes back on.

    •  In operation will transmit from VFO A and receive from VFO B. I’ve had it transmit 40 on meters and receive 2 meters. I’ve never used split before, however it seems “Split” is working just fine.

    • Work around is to ensure VFO A and VFO B frequencies match. That said FLRIG has put your radio into “Split” mode. If you decide to not use the Pi and use Windows, I.E.to use Winlink to send  a E-mail guess   what. Your radio is in split mode. On the FT-857 that does not jump out at you. Leaves you scratching your head trying to figure what up!

    • Issue with Packet, have to close out of everything and then go back it to make it work
.
    • FLRIG just does weird stuff? FLAKY!! 

This may be user error due to that I don’t know how to turn off “Split” and may not have FLRIG configured correctly. Had no problems with “Buster” but FLRIG is a different version with “BullsEye”.

Again had no problems configuring Flrig with “Buster”, no problems with USB ports etc. FLRIG was solid and did not just change things while in use. I have found work a rounds for most stuff to get FLRIG to work in BullsEye however what a pain.

Browser:
When opening Ardop or Packet the browser window may or may not open. Both with Chrome or FireFox.

Conky:
Short Cut from “Ham Radio Menu does not work. However bash ~/pi-build/conky/setconky will launch program. I think the short cut worked one time however not sure.

I know some of this is "in work" and debated if I should send this. On reflection thought there may be some info here that may help with fix.

BAP what a great image!! Thank You Jason.

Barry KD0RQU

Mooman2022, 2022-04-11:
Broke my own rule. Never change more than one thing at a time. Issue number 1 is not BAP or FLRIG issue.

Bought a Deskpi Pi case for my Raspberry 4. With that case you have to add a line of code to make the front USB ports work. That line of code changes something in the boot config that makes it not see the USB port that the radio (CAT) is plugged into when you start FLrig. Unplugging and then plugging the CAT cable back in fixes the problem.  The second thing I did was to do an update. Again .. never change more than one thing at a time. 73 Barry KD0RQU

km4ack, 2022-04-11:
Glad you figured out the issue with the Pi case. Wouldn't have thought that could be the culprit. If I had a dollar for every time I broke that rule and it came back to bite me, I could retire a few years early.

> Conky:
Short Cut from “Ham Radio Menu does not work. However bash ~/pi-build/conky/setconky will launch program. I think the short cut worked one time however not sure.

The shortcut in "Ham Radio" starts conky if it isn't running or restarts it if it is. This is really a left over from when we had issues where Conky wouln't always start at boot. Maybe it should be removed now. 

Set-Conky is there to give the user an easy way to change a few things in Conky without opening the config file.

> Browser:
When opening Ardop or Packet the browser window may or may not open. Both with Chrome or FireFox.

I am seeing the same behavior with Chromium. It hangs whenever we call 127.0.0.1 from the command line. I don't have the issue with firefox. Have you set firefox as default?
`xdg-settings set default-web-browser firefox-esr.desktop`

Mooman2022, 2022-04-14:
Issue with the Pi case is interesting! Knowing the problem is there and having a work around makes it a non issue. (Side note: that Pi case is keeping my Pi 25-35 degrees cooler.)

I have been very lucky and have had no problems with CONKY. I could have sworn that I used the short cut to change the temp. My bad. That said; instead of removing the short cut, link it to Set-Conky?

I can open a Browser.  Such a minor bug.

My main problem is with FLrig putting the radio into split mode. From my research this is not a BAP or BullsEye problem. Some are having the same problem in Windows. So it’s a FLrig problem. For me and the way I operate I find this problem so troublesome I have gone back to Buster to operate. Waiting for a fix from FLrig and I’m sure that Jason will have an update that includes VARA about the same time. <Grin> 

I’m closing this ticket as BAP seems for the most part BAP is working just fine. Thanks Jason for the quick response and for the hard work you put into BAP. It’s awesome! 73

closed 2022-04-14 by Mooman2022